ZoneMinder/zoneminder #912: Implement Passive Source Type for Cams that Support Event Uploading

bhuddah, 2015-06-27:
Maybe someone has an idea how to implement that and we can track it here.
This could be a good idea for larger installations where it is not feasible to analyse all streams on the host.

knnniggett, 2015-06-27:
Not sure what you mean.

bhuddah, 2015-06-27:
i have some cams that support motion detection and upload of videos.
this way you could use zm to only work as some kind of event repository for already identified events and not use the internal logic to process frame by frame data.

knnniggett, 2015-06-27:
This is one of the many features, that would be cool and we are well aware of, but unfortunately it fits into the category of not enough time and manpower to implement.  

I'm going to close this out, not because I don't like the idea (I'd like to get this implemented myself), but because I know this will not be possible to implement in a reasonable amount of time.  If you would like to do the work, then feel free to submit a pull reuqest.

closed 2015-06-27 by knnniggett

kylejohnson, 2015-07-01:
@bhuddah This isn't exactly what you're looking for, but ZM does have a way of being triggered by external cams.  For example, I have an Axis camera that does motion detection, and then alerts ZM to start recording.  This offloads the load onto the individual cameras, and then uses ZM for recording the stream to disk.

See http://www.zoneminder.com/wiki/index.php/How_to_use_your_external_camera%27s_motion_detection_with_ZM and http://www.zoneminder.com/wiki/index.php/AxisMotionDetection