ZoneMinder/zoneminder #914: loop recording

edgiru, 2015-06-30:
Hi!
thank you for this project. you are almost the only ones in the land of the projects brought to the operating state of the video recording to the server.

I would like to ask whether it is possible to set the settings cyclic recording? all the free space on the hard disk ends and you have to clean the files manually.

web2wire, 2015-06-30:
Hi

http://www.zoneminder.com/wiki/index.php/FAQ#How_can_I_stop_ZoneMinder_filling_up_my_disk.3F

Regards

knnniggett, 2015-06-30:
@edgiru Thanks for your support of the ZoneMindner.  Please note that this is a bug reporting forum.  "How do I use ZoneMinder" type of questions should be placed in the user forum at zoneminder.com.

closed 2015-06-30 by knnniggett