ZoneMinder/zoneminder #924: zm_update-1.28.99.sql not executable

hans109h, 2015-07-07:
FYI my db did not update from 1.28.1 to 1.28.99 and I was unable to start up zm due to the IsActive field missing.

I'm not sure but I think this was because the permissions of zm_update-1.28.99.sql were not executable.  After manually updating the db it zm worked.

connortechnology, 2015-07-07:
No, it's because your actual db version is still set to 1.28.1.  This is something I noticed today as well.  
The .sql file does not need to be executable.

hans109h, 2015-07-07:
How is the actual db version set?

ctrlrsf, 2015-07-07:
I'm also not able to start zoneminder service after apt-get upgrade installed 1.28.1+1-vivid-SNAPSHOT2015070301. This is on Ubuntu 15.04.

```
Jul 07 12:29:42 dvr systemd[1]: Starting ZoneMinder CCTV recording and security system...
-- Subject: Unit zoneminder.service has begun start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit zoneminder.service has begun starting up.
Jul 07 12:29:43 dvr zmpkg[4418]: INF [Command: start]
Jul 07 12:29:43 dvr zmpkg.pl[4418]: DBD::mysql::st execute failed: Unknown column 'IsActive' in 'field list' a
Jul 07 12:29:43 dvr zmpkg[4418]: FAT [Can't execute: Unknown column 'IsActive' in 'field list']
Jul 07 12:29:43 dvr systemd[1]: zoneminder.service: control process exited, code=exited status=255
Jul 07 12:29:43 dvr systemd[1]: Failed to start ZoneMinder CCTV recording and security system.
-- Subject: Unit zoneminder.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit zoneminder.service has failed.
--
-- The result is failed.
Jul 07 12:29:43 dvr systemd[1]: Unit zoneminder.service entered failed state.
Jul 07 12:29:43 dvr systemd[1]: zoneminder.service failed.
```

knnniggett, 2015-07-07:
For now, just execute the 1.28.99 sql file manually from the command line.

hans109h, 2015-07-07:
Something else that I noticed was that zm_update-1.28.99.sql was not "installed" to /usr/share... I had to get it out of my build directory.  Is that normal?

ctrlrsf, 2015-07-07:
@hans109h That's strange, on mine it did install to /usr/share/zoneminder/db/zm_update-1.28.99.sql as expected.

hans109h, 2015-07-07:
You're using a PPA, or building from source?

connortechnology, 2015-07-07:
ppa has been updated.

ctrlrsf, 2015-07-07:
@knnniggett Here's what I get when I ran the DB migration manually:

```
# mysql -u root -p zm < /usr/share/zoneminder/db/zm_update-1.28.99.sql
Enter password:
ERROR 1064 (42000) at line 342: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'PRIMARY KEY;ALTER TABLE `States` ADD `Id` int(10) unsigned auto_increment NOT NU' at line 1

# mysql --version
mysql  Ver 14.14 Distrib 5.6.24, for debian-linux-gnu (x86_64) using  EditLine wrapper

# md5sum /usr/share/zoneminder/db/zm_update-1.28.99.sql
b760eaebebb4cb8bbf6f0f14afb900d5  /usr/share/zoneminder/db/zm_update-1.28.99.sql

# dpkg -s zoneminder | grep -i version
Version: 1.28.1+1-vivid-SNAPSHOT2015070301
Config-Version: 1.28.1+1-vivid-SNAPSHOT2015061701
```

connortechnology, 2015-07-07:
there is an error in the db update.  It has already been fixed and a new ppa uploaded.  

You should be on 1.28.1+1-vivid-SNAPSHOT2015070401

ctrlrsf, 2015-07-07:
Hi @connortechnology,

I believe the issue is still present in 2015070401. I just did an apt-get upgrade and it still failed. I also downloaded the raw latest zm_update-1.28.99.sql file from git repo and get the same error.

Here's the full apt-get upgrade output:

```
# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  zoneminder
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 1,082 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu/ vivid/main zoneminder amd64 1.28.1+1-vivid-SNAPSHOT2015070401 [1,082 kB]
Fetched 1,082 kB in 1s (877 kB/s)
(Reading database ... 62314 files and directories currently installed.)
Preparing to unpack .../zoneminder_1.28.1+1-vivid-SNAPSHOT2015070401_amd64.deb ...
Unpacking zoneminder (1.28.1+1-vivid-SNAPSHOT2015070401) over (1.28.1+1-vivid-SNAPSHOT2015070301) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up zoneminder (1.28.1+1-vivid-SNAPSHOT2015070401) ...

Initiating database upgrade to version 1.28.99 from version 1.28.1

Upgrading database to version 1.28.99
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.1 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.99 from 1.28.1
Warning: Using a password on the command line interface can be insecure.
ERROR 1064 (42000) at line 342: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'ALTER TABLE `States` ADD `Id` int(10) unsigned auto_increment NOT NULL PRIMARY K' at line 1
Output:
Command 'mysql -hlocalhost -uX -pX zm < /usr/share/zoneminder/db/zm_update-1.28.99.sql' exited with status: 1
dpkg: error processing package zoneminder (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 zoneminder
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

And here's trying DB migration manually:

```
# md5sum /usr/share/zoneminder/db/zm_update-1.28.99.sql
b6d96934c4d49d6bea32dd7313c0c929  /usr/share/zoneminder/db/zm_update-1.28.99.sql

# curl -O https://raw.githubusercontent.com/ZoneMinder/ZoneMinder/master/db/zm_update-1.28.99.sql     % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 13621  100 13621    0     0  43323      0 --:--:-- --:--:-- --:--:-- 43378

# md5sum zm_update-1.28.99.sql
b6d96934c4d49d6bea32dd7313c0c929  zm_update-1.28.99.sql

# mysql -uroot -p zm < zm_update-1.28.99.sql                                                        Enter password:
ERROR 1064 (42000) at line 342: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'ALTER TABLE `States` ADD `Id` int(10) unsigned auto_increment NOT NULL PRIMARY K' at line 1
```

knnniggett, 2015-07-07:
See #925

connortechnology, 2015-07-07:
Ok, I've uploaded a new PPA.  Should fix the problem.  Let us know.

ctrlrsf, 2015-07-07:
Thanks @connortechnology! Latest 1.28.1+1-vivid-SNAPSHOT2015070402 is :+1:

newtesla, 2015-07-15:
>Database successfully upgraded to version 1.28.99.
>Database upgrade to version 1.28.100 successful.
Works ;) people, this was fast :)

closed 2015-07-17 by SteveGilvarry