ZoneMinder/zoneminder #932: Error "Attempt to directly assign buffer from an undersized buffer of size"

Linwood-F, 2015-07-15:
First time here, hope this is the appropriate way to report an issue. 

There is an off-by-one error in zm_image.cpp.  When the montage view is scaled 1/3 with images that are 2048 x 1536 using remote/rtsp (not sure if the latter matters), I am receiving the error: 

Attempt to directly assign buffer from an undersized buffer of size: 1024650, needed 507x676*3 colours = 1028196

The too-small amount is 506x675x3, i.e. the width and height are off by one. 

In the routine Image::Scale in zm_image.cpp, the scale_buffer_size is calculated by the unrounded multiplication of factor times the original width and height which yields 506 and 675 in this case, times the number of colors, or the too small 1024650.

The routine then loops through width and height with running index last_w_index and last_h_index, yielding a value one larger than the above, which is then assigned to new_width and new_height and used in the AssignDirect call which gives the error .  

I think these new (larger) values are correct, and the buffer size should be recalculated to match, but the buffer size is already used in AllocBuffer.  The simpler solution I think is to do the allocation with +1 in each direction in case you need it yielding either slightly too large, or right size as opposed to sometimes too small..

The " +1" added in the lines below (at the top of Image::Scale) seemed to fix it in some brief testing, but I don't know this code well enough (nor github) to try submitting this as code.

unsigned int new_width = (width*factor)/ZM_SCALE_BASE + 1; // the extra +1 is for rounding errors in the loop, otherwise sometimes the last_x_index can be one more than this
unsigned int new_height = (height*factor)/ZM_SCALE_BASE + 1;   // as above

Hopefully someone who knows this area can see if this makes sense, and is worth fixing.

connortechnology, 2015-07-15:
Thanks for your analysis.  This issue has been around for a while.

I would like to replace this entire function with something more optimized and readable. However for now I think we should just get the fix in.

Interesting reading: http://www.drdobbs.com/image-scaling-with-bresenham/184405045?pgno=1

Linwood-F, 2015-07-15:
Being a photographer I was surprised at the simplicity of how this scaled, and it explained why the scaled images displayed in the browser have so many ragged artifacts at edges. Are the popular image scaling algorithms too slow for use here?   As opposed to reinventing the wheel?

connortechnology, 2015-07-15:
in short yes, but that article I just linked to does a better job.  

I'm also pretty sure that we could leverage some hard based scaling algorithm, or something from an image library instead of rolling our own.

closed 2015-07-15 by connortechnology

connortechnology, 2015-07-15:
btw I think is is also #586

commit 602f703d (2015-07-15): roudn up when calculating buffer size for scaled image. Fixes #932
commit bf9f1835 (2015-07-15): Merge pull request #933 from ZoneMinder/issue_932

roudn up when calculating buffer size for scaled image. Fixes #932