ZoneMinder/zoneminder #935: error starting zoneminder after latest upgrade

newtesla, 2015-07-15:
Starting ZoneMinder: DBD::mysql::st execute failed: Unknown column 'IsActive' in 'field list' at /usr/bin/zmpkg.pl line 318.
failure
 when commented-out lines 310-322, and line 177 - then finally all works, and zoneminder starts.
Setting up zoneminder (1.28.1+1-trusty-SNAPSHOT2015071301)
Ubuntu 14.04 x86, MariaDB mysql  Ver 15.1 Distrib 10.0.20-MariaDB, for debian-linux-gnu (i686) using readline 5.2

connortechnology, 2015-07-15:
I will generate new packages today.  This is something we have been fixing over the last day or two.

knnniggett, 2015-07-15:
@newtesla Did you run zmupdate.pl to update the database schema?

newtesla, 2015-07-15:
Not separately (zmupdate.pl) - all I did was "sudo apt-get upgrade"

Database already at version 1.28.1, update aborted.

knnniggett, 2015-07-15:
You can fix this manually by executing the sql update script, zm_update-1.28.99.sql, from the command line using mysql. Naturally, you'll need to undo the comments you added to zmpkg.pl.

Or you can wait for @connortechnology to build a new package.

connortechnology, 2015-07-15:
new packages have been sent to the ppa.  Will take a little while for ubuntu to build them.

newtesla, 2015-07-15:
>>Database successfully upgraded to version 1.28.99.
>>Database upgrade to version 1.28.100 successful.

Works ;) people, this was fast :)

closed 2015-07-15 by newtesla

davidsmoot, 2015-08-21:
I have exactly the same error message from exactly the same root cause.  I am running ubuntu 14.04 and did a "sudo apt-get upgrade" last night (Wednesday 2015-08-19).  This morning I noticed zoneminder was broken.  Another apt-get upgrade and reboot did not improve things.  I tried a "sudo service start zoneminder" and got exactly the same error as the original poster.  Looking in usr/share/zoneminder/db/, I have many zm_update-1.28.xx.sql including the .99 referenced here.  Should I apply them all in sequence, should I apply just the 99?  Thanks for your time.  
David

Linwood-F, 2015-08-21:
I think you can just run zmupdate.pl and it figures it all out for you.

davidsmoot, 2015-08-21:
Thank you for the prompt reply but unfortunately, I think I am in more trouble now...

I executed "sudo zmupdate.pl" and here is my output:
Initiating database upgrade to version 1.28.104 from version 1.28.1

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop : 

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.28.104
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.1 from 1.28.1

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.100 from 1.28.1

Database successfully upgraded to version 1.28.100.
Upgrading DB to 1.28.101 from 1.28.1

Database successfully upgraded to version 1.28.101.
Upgrading DB to 1.28.102 from 1.28.1

Database successfully upgraded to version 1.28.102.
Upgrading DB to 1.28.103 from 1.28.1

Database successfully upgraded to version 1.28.103.
Upgrading DB to 1.28.104 from 1.28.1

Database successfully upgraded to version 1.28.104.
Upgrading DB to 1.28.99 from 1.28.1
ERROR 1142 (42000) at line 376: CREATE command denied to user 'zmuser'@'localhost' for table 'Servers'
Output: 
Command 'mysql -hlocalhost -uzmuser -pzmpass zm < /usr/share/zoneminder/db/zm_update-1.28.99.sql' exited with status: 1

 Correct me if I am wrong, but it appears that the scripts were applied out of order?  It appears to have tried the 99 script after all the 100 series scripts?  

Just for fun I tried to start zoneminder again and now I just get Starting ZoneMinder: failure

Again, thanks for trying to help.

Linwood-F, 2015-08-21:
Looks like it went in alphabetical order. Maybe no one ever went about 100 before?  Not sure.  I've only gone 1 or 2 at a time myself.  Hopefully someone will come along and recommend something, but if you were at .1, it may be worth manually running each in (correct) sequence and see what happens.  

I won't mention the answer above at "backup'.  :(

SteveGilvarry, 2015-08-21:
Zmuser needs create permissions for 1.28.99.

SteveGilvarry, 2015-08-21:
But also correct that it is not applying them in the correct order, but the actual error is from this line

"CREATE TABLE `Servers` (
  `Id` int(10) unsigned NOT NULL auto_increment,
  `Name` varchar(64) NOT NULL default '',
  `State_Id`    int(10) unsigned,
  PRIMARY KEY (`Id`)

From Debian readme
grant lock tables, create, alter on zm.* to 'zmuser'@localhost identified by "zmpass";

We should look at permissions that zmupdate runs under.

davidsmoot, 2015-08-21:
Thank you both.  I'm not going to be able to finish this tonight but I will tackle in the morning.  If I understand correctly:
log into mysql from the command line
issue the command to give permissions above
re-run zmupdate.pl

davidsmoot, 2015-08-22:
Thank you gentlemen, you solved my problem.

I issued the command given by SteveGilvarry to get the permissions updated and then I re-issued the zmupdate.pl command and it succeeded.  I then issued "sudo service zoneminder start" and that succeeded.

I am somewhat confused that the web interface appears to have changed URL but that is a separate thread and a separate Google.

kelawargila, 2016-12-27:
Dear all,
i did an apt-get dist-upgrade (from Debian Wheezy to Jessie).  my zoneminder is no longer functioning. when i run the command 
      zmupdate.pl
this is the error message that appears : 

root@cockroach:/home/insect-king# zmupdate.pl

Previous versions of ZoneMinder used the MyISAM database engine.
However, the recommended database engine is InnoDB.

Hint: InnoDB tables are much less likely to be corrupted during an unclean shutdown.

Press 'y' to convert your tables to InnoDB or 'n' to skip : y

Converting MyISAM tables to InnoDB. Please wait.
DBD::mysql::st execute failed: Can't find file: 'Stats' (errno: 2) at /usr/bin/zmupdate.pl line 365, <STDIN> line 1.
Can't execute: Can't find file: 'Stats' (errno: 2) at /usr/bin/zmupdate.pl line 365, <STDIN> line 1.

would any be able to help me on this?  thanx very much in advance.

knnniggett, 2016-12-27:
The root cause of this problem was fixed back in 2015. If you are still experiencing a problem with zoneminder, it is not related to this particular issue. Please report it in the user forum.