ZoneMinder/zoneminder #939: zmtrigger: minor message format correction

onlyjob, 2015-07-16:
[AxisMotionDetection](http://www.zoneminder.com/wiki/index.php/AxisMotionDetection) wiki page describes a nice method to offload motion detection to the camera capable of triggering motion event in Zoneminder.

Unfortunately I run into problem with `+` character described on the above page: even with latest firmware CGI code of the _AXIS 210_ camera filters out `+` from the message. Therefore camera is unable to send the following message:

    3|on+11|1|motion|text|showtext

but the this message (where `+` replaced with `_`) is OK:

    3|on_11|1|motion|text|showtext

I suggest we modify zmtrigger.pl [line 449](../blob/master/scripts/zmtrigger.pl.in#L449) with the following change:

~~~~
- elsif ( $action =~ /^(on|off)(?:\+(\d+))?$/ )
+ elsif ( $action =~ /^(on|off)(?:[+_](\d+))?$/ )
~~~~

To allow character `_` as alternative to `+`.
I shall be happy to prepare pull request if everybody is all right with such change.

Thanks.

knnniggett, 2015-07-16:
Probably.  
But can you compare what you've written here to this?
http://www.zoneminder.com/wiki/index.php/AxisMotionDetection

connortechnology, 2015-07-16:
It would be nice if non of this was necessary.  

I have 212PTZ cameras and their firmware stops at 4.49 and they have this problem with the +.  I say we accept a space as detailed in http://www.zoneminder.com/wiki/index.php/AxisMotionDetection

closed 2015-07-16 by knnniggett

onlyjob, 2015-07-17:
@knnniggett, Of course I already digested wiki page and it is even mentioned in my bug report...

@connortechnology, I appreciate your quick fix but I thought we are still discussing it?
Personally I do not like space(s) in the trigger protocol (it feels like bad semantics).
Spaces should not be significant and when spaces are matched by regex it is better to use `\s+` to allow one or more character. IMHO underscore would be better...

Thank you both for quick reply. :)

connortechnology, 2015-07-17:
My reason for going with a space is that for people with this problem, following the docs will enter a + but the camera will silently replace that + with a space.  So with this change, it will simply work and they won't bother us.

onlyjob, 2015-07-17:
Good point. We just need to remember to update docs and improve regex as well (since it is a bad practice to use white spaces in regexes (`/x` mode compatibility etc.)).

PR #941 (2015-07-16): add space instead of + to handle old Axis cameras
fixes #939