ZoneMinder/zoneminder #940: Montage shows only feed from the first camera

onlyjob, 2015-07-16:
I have two cameras and they are both working fine in Zoneminder.
Individually I can open each camera's view and see live feed etc.
However in "montage" view only images from first camera are shown -- second camera frame remains empty/blank even when I change order of cameras.
When I open "montage" window `zm_monitor.cpp/3893` logs

    zms | ERR | Unable to send stream frame: Broken pipe

knnniggett, 2015-07-16:
That message is a nuisance alarm that sometimes occurs when one simply closes a live view.  It could also mean that the network connection from server to client dropped.... not sure that is helpful in this context.  What else do you see in the logs?

Things to check:
1) What browser are you using?  Firefox is default limited to 6 streams.  If Firefox has not yet closed out previous streams (this was a past issue) then Firefox could be counting those as part of you 6 streams.

2) What does the load look like on your machine?  It is hard to imagine just two cameras would overload it, but it you are running in a very limited environment with high frame rate and/or resolution then it is possible.

3) Can you open both cameras separately?  If not, then you may be looking at a ram issue.  ZoneMinder needs room in /dev/shm for mmap files.

onlyjob, 2015-07-17:
There are only two more messages in log:

 * zms: "Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results for monitor NNN"
 * web_php: "socket_sendto( /var/run/zm/zms-375646s.sock ) failed: No such file or directory"

1) I tried Icedove (a.k.a. Firefox) and Chromium: both affected by exactly the same problem.

2) Load is within 25%, all 4 CPUs are not too active...

3) I can't. Second camera window is blank,and it doesn't matter which one is opened first. Looks like the same problem. However I doubt it could be lack of RAM as `df -h` show

~~~~
Filesystem             Type      Size  Used Avail Use% Mounted on
tmpfs                  tmpfs      24G  101M   24G   1% /dev/shm
tmpfs                  tmpfs      24G  1.2G   23G   5% /tmp
~~~~

onlyjob, 2015-07-17:
Interesting that when I open first camera window (working) the following message appears in log:

    zms[17932]: WAR [Shared data not initialised by capture daemon, some query functions may not be available or produce invalid results for monitor NNN]

However opening second camera window (not working) show errors/warnings (that I mentioned earlier) instead...

onlyjob, 2015-07-17:
Another thing: as soon as I close first camera window, second window starts showing live feed...

onlyjob, 2015-07-17:
I have this very problem on "master" as well...

knnniggett, 2015-07-17:
Have to think about this one. You aren't running this inside a Docker container or anything else that might artificially restrict the amount of access to /dev/shm?

connortechnology, 2015-07-17:
That message generally means that zmc isn't running.  So why isn't it running?  Look at your logs. if there isn't anything there, turn on debug logging.  We are not going to be able to help without information.

onlyjob, 2015-07-17:
Thanks guys. I already tried debug logs but couldn't find anything relevant. There are  `/dev/shm/zm.mmap.*` files so it is very strange that only camera can stream at a time...
I'm running Debian "testing"; no Docker containers or funny stuff...
At the moment I'm lost without ideas what to try next...

connortechnology, 2015-07-17:
before viewing montage, do ps ax | grep zmc.  Do you see a zmc process for each monitor?

In your error line, you have for monitor NNN I assume thats your replacement.  If it actually says NNN then that would be interesting.

please post logs from zmc and zms with debug level at least 3.

onlyjob, 2015-07-17:
Yes, both `zmc` processes are there. I think I found the problem in my _nginx/fcgiwrap_ configuration. It actually works in _Apache2_ because the latter can spawn two CGI processes. It seems that _fcgiwrap_ just uses one process and communicate through one socket which is insufficient in my case... I'll investigate...

connortechnology, 2015-07-17:
yeah that would do it.

onlyjob, 2015-07-17:
There were two problems:

 1. I didn't know that with `fcgiwrap` I need to pre-fork as many processes as I have cameras.
 2. The actual bug in `fcgiwrap.service` ignoring `DAEMON_OPTS` from `/etc/default/fcgiwrap` file (on systemd). See corresponding bug:
  http://bugs.debian.org/cgi-bin/bugreport.cgi?bug=792705

closed 2015-07-17 by onlyjob