ZoneMinder/zoneminder #948: PHP warnings

onlyjob, 2015-07-17:
"master" exhibit the following warning

1) on top of `index.php`:

    Warning: implode(): Argument must be an array in /usr/share/zoneminder/www/skins/classic/views/console.php on line 74

2) on top of `index.php?view=monitor&mid=1`:

    Notice: Use of undefined constant ZM_SERVER_ID - assumed 'ZM_SERVER_ID' in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 43

In "options" there are no servers under "Servers" tab, but "AddNewServer" button show empty page with error "View 'server' does not exist".

connortechnology, 2015-07-30:
I believe I fixed this... can you confirm with latest master and close if so?

onlyjob, 2015-08-01:
1) is still there.

2) is gone. (Thank you).

onlyjob, 2015-08-03:
Just noticed another warning: `Notice: Undefined index: Id in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 116` in "Add New Monitor" dialog...

commit dff624a4 (2015-08-11): Merge pull request #985 from ZoneMinder/fix_state_load

fix #948 1
commit e178e1b1 (2015-08-11): fix #948 1
closed 2015-08-11 by connortechnology