ZoneMinder/zoneminder #958: Build with cmake add cpp file names with complete build directory to system logs

manupap1, 2015-07-22:
Reason is that cmake defines `__FILE__` preprocessor macro with absolute path unlike autotools.

This issue has already been experienced by others (http://www.cmake.org/pipermail/cmake/2013-January/053114.html) and it seems there is no simple fix except the hack suggested in the thread (http://www.cmake.org/pipermail/cmake/2013-January/053117.html).

This hack would consist in overriding `__FILE__` macro in CMakeLists.txt with:
```cmake
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D__FILE__='\"$(subst ${CMAKE_SOURCE_DIR}/,,$(abspath $<))\"'")
```
Warnings about redefined builtin macro can be removed by adding `-Wno-builtin-macro-redefined` to CXXFLAGS.

The other option would be to improve relaled functions in `src/zm_logger.h` but I don't think an isssue with a build tool should lead to modifications in source code.

knnniggett, 2015-07-28:
When I view the zoneminder logs on my system (rpm package built using cmake), the File column is populated similar to this:
`/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_monitor.cpp`

This contains the absolute path to the directory zoneminder was build from.

Are you saying you see something different?  
If so, perhaps I can dig into my rpm build environment and figure out how that is being set.

If this is not what you mean, then apparently I don't understand the question.

manupap1, 2015-07-28:
This is what I mean.
With autotools, the File column is populated only with file names.

knnniggett, 2015-07-28:
I see; I misread the title.  I thought you wanted to add that in there.  I'll look into it.

knnniggett, 2015-08-21:
I've read the arguments in the references from the initial post, and I have chosen to fix this by modifying logger.cpp.  Doing it this way guarantees we can fix it once and not have to return to this again.  In any case, I'll leave it up to my peers to agree or disagree with this decision.

Note that logging from Perl and PHP scripts is unaffected by this.  PHP still reports the full (valid) path to the file in question and I think Perl just reports the filename.

closed 2015-08-22 by connortechnology

gourav5, 2017-11-27:
provide me the path of given below file.

zm_monitor.cpp

ubuntu 16.04 server

PR #1029 (2015-08-22): Remove full path from Logger filename
fixes #958 

Only affects logging from cpp.  Does not affect Perl & PHP scripts.