ZoneMinder/zoneminder #960: Add Montage-like history review screen

Linwood-F, 2015-07-22:
This is a proposal to add a new feature which is a combination of the timeline and the montage screens.  It would display historical data for all (viewable) monitors at one time, while building a timeline showing the availability of images and alarms similar tot he timeline graph, but using a heatmap format instead of bar graph (i.e. darker colors indicate alarms). 

There are several goals of this: 

- To animate all monitors as you scrub through the timeline

- To have a "play" mode that also plays at various speeds all monitors at once, running when you are not scrubbing through.

- To be "responsive", in the sense that browser size and resizing does reasonable things rather than running off the screen.

- To use more modern HTML constructs that are more likely to work across browsers (as opposed to the very densely backed divs in the timeline that are not handled well in some browsers).

Some non-goals I propose: 

- Not to support legacy browsers; html5 support is required.

- Not to support low bandwidth or weak processor clients (this moves and displays a lot of data, and there is really no way around that with the present architecture that I see).

- No Full fps video -- the displays are aimed at 1 second real time intervals as a reasonable compromise of animation vs. performance.

I am also trying to make it support touch, but that is not a priority as I do not see this as getting much mobile interest due to bandwidth and CPU horsepower requirements.

I have a sample code done, but not linked in, that I will put up as soon as I can figure out how to do so.

Usage notes: 

- It is NOT linked to the event display like the timeline, rather it has selection for time frames as buttons, and as zoom and pan.  Zoom is centered on the slider.

- It DOES link out to the existing event replay option based on the time of the current display - click on the monitor (not the timeline) to link to it.

- To reposition the timeline scrub slider, click on the timeline.  To slide it click and hold.  (I was frustrated with the existing timeline that any mouse-over affected it even if you did not want to). 

- Until it is integrated on the console, you can invoke it with a standard view call.  Drop it into the same location as timeline.php, and run with a URI of /index,php?view=montagereview

Question: Do you think this should go as an additional option where the "montage" option is now, e.g. "cycle / montage / history montage"?

Linwood-F, 2015-07-22:
I have, perhaps, if I did it correctly, put in a branch named 960-montage-review-new-feature which contains the code for this.  No pull request as I would really appreciate feedback so I can finish it up.

knnniggett, 2015-07-23:
You need to trust me on this and generate a pull request sooner rather than later.  When a pr is unfinished, all you have to do is say so (which you have done), and we will label it not ready to merge.  When you generate the pull request, it will highlight everything you've done, making it very easy for anyone to see your changes.

By waiting you are making it harder as your branch falls behinds newer changes that get added into master.

SteveGilvarry, 2015-07-23:
You will end up with something like this.
https://github.com/ZoneMinder/ZoneMinder/compare/master...Linwood-F:960-montage-review-new-feature

Linwood-F, 2015-07-23:
Let me incorporate the latest feedback from a discussion on irc and I'll do the PR.

The biggest unknown to me is the proper place to put it.  I am not positioning it to replace the existing timeline as frankly I could not unravel all the different ways (parameters) that it could be called with.  I think some of the code there is dead and unreachable but I could not be sure.  And frankly for me personally the link to specific event display pages is more restrictive than helpful, I just wanted "show me time and let me browse".   So in the current commit I put it on the console, but a couple people commented that was the wrong place and/or name (without a very clear idea where the right place was).   So was hoping more feedback there.

Plus a few bugs to fix.  Sooner rather than later though, will.

Linwood-F, 2015-07-24:
Note: there is I think one line in here that uses the mootools.js.  They appeared to be used already, but then I saw that they were in the angular-js branch and not sure how to tell if they are available in the master branch.  It's just date formatting so if those tools are not part of the core product can someone let me know?

knnniggett, 2015-07-24:
mootools is in master:
https://github.com/ZoneMinder/ZoneMinder/tree/master/web/tools/mootools

Linwood-F, 2015-07-27:
I assume this is obvious - I'd love feedback, but I want to work on this some more before it is incorporated (if).   It has a couple minor bugs relating to rounding of time, which sometimes leaves 1 second gaps which flash whitespace, for example.  But mostly I want to work on different formatting.

knnniggett, 2016-03-12:
I just realized this is still open and will close it now since #961 has been merged.
Go claim your bounty @Linwood-F !

closed 2016-03-12 by knnniggett

Linwood-F, 2016-03-12:
I had no idea.  Interesting.  Will go figure it out.  Still want to do a V2 with some improvements but cannot seem to get back to it.

connortechnology, 2019-01-08:
@Linwood-F did you end up claiming this bounty?

Linwood-F, 2019-01-08:
@connortechnology Yes, I claimed this one on 3/12/16 and donated it back to increase the bounty on "Store events in MP4 container".