ZoneMinder/zoneminder #970: Change of scale on live view (only) does not re-scale image just img tag

Linwood-F, 2015-07-27:
Reproduce Scenario: 

Set the scale for the live view to something less than Actual (e.g. 25%) in the options.

Go into the live view - image appears properly at 25%.

Change the scale on the page to Actual.  Image size changes, but resolution does not - image is very low resolution.

The cause of this is due to the change_scale javascript changing only the width/height of the img tag, but not changing the &scale=25 that is in the URL fed to nph-zms, so the streaming server continues to stream at 25%, but the browser is upscaling it.

I suspect it is necessary to stop and restart the stream (maybe this happens anyway), have not looked.  I THOUGH this worked properly before, but I see no changes that might have caused this, so I suspect it is an old issue.  I still don't really understand zms so haven't tried to change it myself (yet, will if I can).

Linwood-F, 2015-08-05:
I tracked this down.  It is not a new regression, but an old change.

https://github.com/ZoneMinder/ZoneMinder/commit/09bf78f865313abc9ed49bec703ef9a1d9e67729

The comment says "worked in FF2 but kills FF3" and dates from 2009.  As we are at FF39 now, I suspect putting it back would be OK, HOWEVER...

I tested putting it back in both IE and Chrome without success, it appears the change of the URL is not terminating the old-scale screen. I didn't dig deep enough to know how to do this properly, the continuous stream stuff is still a bit of a mystery to me.

It's easy to reproduce -- set your options default to 25%.  Go to the console and click a monitor, it opens at 25%.  Change on that screen to Actual, and you'll see you get a very poor quality image, it is scaling the image tag but not the stream's contents.

SteveGilvarry, 2017-03-18:
Review and Action

closed 2018-06-22 by stale[bot]