ZoneMinder/zoneminder #977: ffmpeg blue screen high CPU usage on a inaccessible camera

qoole, 2015-07-29:
Hi all,

I've just upgraded to latest Zoneminder (v1.28.100 according to the webui).
```
ffmpeg version 2.7.2 Copyright (c) 2000-2015 the FFmpeg developers
  built with gcc 5.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-shared --enable-avisynth --enable-avresample --enable-decoder=atrac3 --enable-decoder=atrac3p --enable-dxva2 --enable-fontconfig --enable-frei0r --enable-gnutls --enable-gpl --enable-gray --enable-ladspa --enable-libaacplus --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcelt --enable-libdc1394 --enable-libfaac --enable-libfdk-aac --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libiec61883 --enable-libmodplug --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopencv --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libutvideo --enable-libv4l2 --enable-libvidstab --enable-libvo-aacenc --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-nonfree --enable-openal --enable-opencl --enable-opengl --enable-openssl --enable-runtime-cpudetect --enable-swresample --enable-vdpau --enable-version3 --enable-x11grab
  libavutil      54. 27.100 / 54. 27.100
  libavcodec     56. 41.100 / 56. 41.100
  libavformat    56. 36.100 / 56. 36.100
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 16.101 /  5. 16.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.100 /  1.  2.100
  libpostproc    53.  3.100 / 53.  3.100
```

I have a couple of cameras from three different manufacturers that are currently inaccessible over the network due to network upgrades taking place. This hasn't been a problem until the update. Now, however the 'Source' entry for the cameras stays green (as if it's successfully connected) and when I click on the stream for the video I get a blue screen. If check top the zmc for each of these cameras is using 70%+ CPU (probably all CPU that is available to them).

Short of disabling these cameras (prone to me forgetting to re-enable them) is there any options I can set to make ffmpeg fail like it used to when it can't connect rather than showing that blue screen.

I do not want to use the native RTSP I would like to continue using ffmpeg.

Thanks,

Alex

knnniggett, 2015-07-29:
This is a known issue.
Place a value in the MAX fps field of each monitor to a value that is just higher than the frame rate in the camera. It is important that this value is larger than the camera's fps, never less than.
This will keep your cpu from pegging should the camera drop off the network.

qoole, 2015-08-01:
Ah ha, suitable workaround!

Will this issue be kept up to date when a fix is found?

knnniggett, 2015-08-01:
This issue has been discussed multiple times in the user forum.
You should follow the user forum and the release notes of future releases to stay on top of this issue.

closed 2015-08-01 by knnniggett

qoole, 2015-08-02:
....then what's the point of an issue tracker....

How can you close an issue that is still an issue?

I searched the issue tracker to find out if anyone else had this problem, couldn't find anything so that's why I posted.

Why not leave it open until a fix is made that way people can easily find out that it is a known issue.

knnniggett, 2015-08-02:
This issue is discussed repeatedly both in our github forum and in the user forum. Try searching on  "maxfps". Creating new issues in our githb forum about known issues makes our job harder.  

We know about this, but we do not have the resources to make this a priority.  You can, of course, submit a pull request yourself.

Please review our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

When in doubt, ask us in the IRC channel.