ZoneMinder/zoneminder #979: Unable to decode jpeg / unable to allocate buffer for large JPEG

onlyjob, 2015-07-31:
I'm trying to pull one frame per minute from the following URL:

  * http://www.weathercamnetwork.com.au/new_farm/982x737.jpg?cache=

I've adjusted image dimensions to 640x480 but `zmc` reports the following errors:

| message | File | Line | 
| ---- | ---- | ---- |
| Unable to decode jpeg | src/zm_remote_camera_http.cpp | 1151 |
| Failed requesting writeable buffer for reading JPEG image. | src/zm_image.cpp | 1011 |
| Held buffer is undersized for requested buffer | src/zm_image.cpp | 330 |

Reproduced on recent "master" as of 7...14 days ago.

Where can I find the buffer settings?
Can ZoneMinder process large JPEGs such as this one?
Any ideas how to fix this problem please?
Thanks.

connortechnology, 2015-07-31:
The jpg you are requesting is 982x737... So your 640x480 buffer size is too small.

Try setting the image dimensions to 982x737.

onlyjob, 2015-07-31:
I tried smaller dimensions because of `Image size is not multiples of 12 and 16` reported by `/src/zm_camera.cpp` at line 45. If I set image dimensions to 1024x768 I get `	Captured image 2171202 does not match expected size 2359296 check width, height and colour depth` in `src/zm_monitor.cpp` line 2892...

onlyjob, 2015-07-31:
It is like _catch-22_: the original dimensions does not satisfy _multiples of 12 and 16_ rule; the closest to acceptable dimensions are 992x744 or 992x752 both produce `does not match expected size` error...

connortechnology, 2015-07-31:
patch on the way.

after patch 982x738 will work.

closed 2015-07-31 by knnniggett

onlyjob, 2015-08-01:
Thank you, it works with 982x744 (Y needs to be a multiply of 12).

onlyjob, 2015-08-01:
Actually something is still not right. Although I can see image in camera window there are no frames getting saved in "Record" mode and there are zero events for this camera. Could you try with URL I provided please?

onlyjob, 2015-08-01:
Never mind, Events are created properly it just took longer than I expected. All good -- thank you very much for a quick fix. :)

PR #980 (2015-07-31): Fix image dimensions check
Fixes #979 

I thought I got this fix in a long time ago.  Must have gotten lost along the way.