ZoneMinder/zoneminder #981: ..

tinygoldenbooks, 2015-08-03:


knnniggett, 2015-08-06:
closing due to lack of content

closed 2015-08-06 by knnniggett

tinygoldenbooks, 2015-08-06:
Hello,

 Didn't know that I opened an issue when I donated through bountysource.
Is there a better way to donate?

I apologize for the inconvenience.

Thanks,

Nelson

knnniggett, 2015-08-06:
Bountysource is the current method for donating.  Thank you.

It is good to know how these empty issues are getting created.  I think one other person did the same thing a while back. It doesn't happen to most who donate.  It must be something during the bountysource process that does this.  Not sure what exactly.

tinygoldenbooks, 2015-08-06:
Thanks for the reply,

 I saw that that bounty source allows automatic monthly donations after
poking around on their site for a few minutes. I decided to go that route.
I know it's only 10 bucks a month, but I really enjoy what you make and I
can't program so I figured this would help.

Keep up the awesome work,

Nelson