ZoneMinder/zoneminder #987: Feature Request - Filter by zone name

brendonanderson, 2015-08-04:
I have a camera that has two zones: one for the street and one for the walkway leading up to the house.  I would like to get an email when the zone for the walkway is triggered, but not when the street triggers (though I would still like to record both events).

knnniggett, 2015-08-06:
This does not meet our requirements for creating an issue in our github forum.  

Criteria has been documented here:
http://www.zoneminder.com/wiki/index.php/Documentation

Feel free to reopen this issue if and when you are prepared to meet the requirements.

If you want to drum up support for this idea, feel free to discuss it with us directly in our IRC channel or with our user group in the user forum.

closed 2015-08-06 by knnniggett

koverly, 2015-08-12:
I looked at the referenced link and did not see the criteria for creating an issue.

I would also like to see the same functionality indicated in the post.

knnniggett, 2015-08-12:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Original post has been updated.  The link is also shown in Readme.md, on the main github page.