ZoneMinder/zoneminder #988: zmwatch.pl consumes ram

connortechnology, 2015-08-04:
after running for a while, zmwatch consumes lots of ram.  This is common among perl programs.  

There may be a memory leak somewhere...

Proposed solution is to just restart it nightly.

onlyjob, 2015-08-07:
I have different impression about Perl programs and I am unable to confirm this problem... Even after long run `zmwatch.pl` do not leak memory on my Debian system...
IMHO there is no need to restart -- at least not by default...

alager, 2015-10-07:
I've been watching zmwatch.pl using top, continuously for the past week, and I do not see it growing at all.  It's been consistently at these numbers:
```
  VIRT      RES     SHR    %MEM
143324     14492    5296    0.1
```
I've got 5 cameras, all doing 10fps, 3 with motion zones, averaging about 3k events over 5 days.

closed 2017-03-18 by SteveGilvarry