ZoneMinder/zoneminder #991: Add option to make TimeStamp larger

schrorg, 2015-08-06:
I am working on a patch to double-size the timestamp in pictures (for better readability in scaled versions). I therefore created a "zm_bigfont.h" with a larger font and did a few modifications in zm_image.h and zm_image.cpp. Everything seems to work now, but I would like to make the size configurable for either every single monitor or the whole system.

Are there any concerns in making this an option? And would you prefer to have that a global option or an option per monitor?

schrorg, 2015-08-06:
Everything I've done until now is in https://github.com/schrorg/ZoneMinder/tree/large-timestamp

knnniggett, 2015-08-06:
We need you to take this a step further and generate a pull request.  
That is not a problem if you feel it is not quite ready.  Simply state that in the description.  
This makes it easier for us by highlighting exactly what you changed, and it makes it easier for you by avoiding future merges to your branch as new commits are added to the master branch.

pliablepixels, 2015-08-06:
@schrorg I looked at your changes. It looks like your code is basically replicating pixel rows of the font as ZM does not really use a font vector, it defines a bitmap for each character in an array. What is your thought on readability? While it does increase the size, do you find the resulting text is blocky and hard to read? IF so, would it make sense to include an option to overlay a vector font that is royalty free? (Any google font can be used as a local font file and is license compatible afaik)

schrorg, 2015-08-07:
@pliablepixels I created the big font data array by first duplicating every bit (pixel) horizontally and then duplicating each line. So basically, I made four pixels out of one. I did this with a few lines of perl. But the effect is a bit like if you would view a video in 2x scale:
[image]
It is absolutely okay for us but sure could be better.

I will see and talk to some colleagues about a vector font. It seems to me that the piece of code should be kept as fast as possible and I have no idea about how changes in that area would impact the image processing performance.

pliablepixels, 2015-08-07:
Thanks - it looks much better than the tiny display today. I'd recommend you do a PR as @knnniggett suggests with this enabled. You should also look at enabling this option via the web console otherwise people won't be able to use it.

schrorg, 2015-08-08:
@pliablepixels I did a pull request (#992) and I am working on an option for the web interface.

pliablepixels, 2015-08-08:
@schrorg - nice! thanks.

knnniggett, 2015-08-18:
#992 was merged. Closing #991 .

closed 2015-08-18 by knnniggett