ZoneMinder/zoneminder #999: Malicious tracking code within vjs.zencdn.net video.js

wdeninger, 2015-08-08:
Is there a strong reason why zoneminder in the video writer branch needs to use the video.js supplied by vjs.zencdn.net?    I'm rather concerned about the security issues arising from needing to 'trust' that zencdn will not push malicious code through my firewall to my LAN clients, not to mention all the goggle-analytics tracking that seems to be attached to it.  Incidentally, according to a quick google search, the video.js from zencdn was responsible for spreading a malicious password stealing Trojan (http://blog.videojs.com/post/61309840958/unauthorized-modification-of-videojs-cdn-files). 

-William

./web/skins/classic/views/event.php:<link href="//vjs.zencdn.net/4.11/video-js.css" rel="stylesheet">
./web/skins/classic/views/event.php:<script src="//vjs.zencdn.net/4.11/video.js"></script>

The above code allows zencdn.net to to executable javascript pulled in from their servers which track zoneminder video request including but not limited to your hostname, path, machine type, etc. 

... a cursory inspection of video-js as of 8/8/2015:

!function(t,a,e,n,m){m=a.location,t.src="//www.google-analytics.com/__utm.gif?utmwv=5.4.2&utmac=UA-16505296-2&utmn=1&utmhn="+n(m.hostname)+"&utmsr="+a.screen.availWidth+"x"+a.screen.availHeight+"&utmul="+(e.language||e.userLanguage||"").toLowerCase()+"&utmr="+n(m.href)+"&utmp="+n(m.hostname+m.pathname)+"&utmcc=__utma%3D1."+Math.floor(1e10*Math.random())+".1.1.1.1%3B"+"&utme=8(vjsv)9(v4.11.4)"}(new Image,window,navigator,encodeURIComponent);

SteveGilvarry, 2015-08-08:
Nope was fastest option for a quick test of videojs, feel free to PR a
change. At time was not sure of how to handle new js file in auto tools.
Would have been changed before merging as Debian not fans of CDN versions.

SteveGilvarry, 2015-08-08:
FYI closing this as not a bug in current master.

closed 2015-08-08 by SteveGilvarry