ZoneMinder/zoneminder #1000: Add optional scale/width/height to query in image.php

Linwood-F, 2015-08-09:
The image.php is used to return a single historical image.  It is currently an alternative to direct access via the /events... path, the latter of which is a security issue and seems doomed to going away.  The image.php is an alternative, though in the future I am a strong proponent of incorporating historical access to single frames in zms instead.  However that is a large task, and image.php is already used and simple, and more relevant is being incorporate in new programs.

This issue is to incorporate scaling as an option in this program.  There is a tradeoff in doing so, in that scaling on the server puts load there, but removes potentially very large bandwidth requirements if clients pull data with an explicit scale (as opposed to pulling the whole image and then scaling in the client).

pliablepixels, 2015-08-09:
This is a good feature to have. zmNinja uses view=image to apply authentication over event frames without the (current) overheads of zms. Till the time zms is fixed to a) return a single image for an event b) validated that it does not have any significant overhead compared to this method, I personally don't see any reason why this new addition should be a problem for what it does.
Hopefully, this will be accepted!

Linwood-F, 2015-08-09:
Thanks, I've pushed a change to permit either width or height or scale in combinations.

pliablepixels, 2015-08-10:
Seems to work fine for me. @Linwood-F you can always make (optional) changes to the algorithm in later pushes if you want, but its useful to have as it stands today. Would you want to request folks to merge it?

Linwood-F, 2015-08-10:
Yeah, as you saw offline I experimented with the other techniques and did not see improvements (and one of them crashes php consistently), so i think this is ready, will post a note in the pull request.

knnniggett, 2015-08-12:
PR was merged.  Can we close this issue out?

Linwood-F, 2015-08-12:
Sure, sorry, still learning the steps.

closed 2015-08-12 by Linwood-F

knnniggett, 2015-08-13:
No problem.  Thank you for all the work you've done for the project.