ZoneMinder/zoneminder #1011: zmc when only monitor

speeskeek, 2015-08-13:
Hi there,
I've been using ZM now for a while, and switched recently from motion detection on my images to 24/7 recording.

So I have sub-streams (CIF, 2fps, 8-bit gray) that I record (function=record)
Ans I have the main-streams (704x578, 5fps, 24-colour) to look at when I like to (function=monitor)

But what I don't get is that zoneminder is capturing the main streams, even when I'm not looking. This is causing zmc to consune cpu usage, when it is not neecssairy. Is this normal (is it a bug? no its a feature), or is there something wong with my config?

What looks more normal to me is that zmc starts capturing images when I want to live-view te stream (so feeds them to nph-zms).

Any toughts?

PS: I tried to ask this on the regulat forum, but the captya is bullying me, can't get past it :)

knnniggett, 2015-08-13:
This does not meet our posting requirements to the github forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-08-13 by knnniggett

speeskeek, 2015-08-13:
Running:
ZM 1.28.1
Debian 
3.16.0 kernell 
amd64
3 IP camera's via ffmpeg (rtsp link from the camera)