ZoneMinder/zoneminder #1013: Migration of material in ZM Wiki to Read the docs

pliablepixels, 2015-08-13:
As of today, http://www.zoneminder.com/wiki/index.php/Main_Documentation and http://zoneminder.readthedocs.org/en/latest/ are not in sync. It would be good to have a single repository of information so they don't keep going out of sync, which is one big issue with ZM.

Per my discussion a few months ago with @SteveGilvarry all formal documentation  goes to readthedocs and 3rd party contributions not officially sanctioned stay on the wiki (for example most of http://www.zoneminder.com/wiki/index.php/Contents) 

I'll start moving the rest of the stuff that @SteveGilvarry started with and will keep checking with the devs in the interim to make sure I am importing the right stuff

pliablepixels, 2015-08-15:
@connortechnology while updating the ubuntu documents I noticed all guides recommend that we add a 15 second sleep timer to the startup script so ZM starts after mysql is running

a) Why not just add that echo "Waiting..." and sleep 10 in the packaged PPA ?

b) In general, I find this to be a little suboptimal. IF you need to start ZM, you are always waiting around for 10 seconds even if mysqld is running. I use this script which works reliably for me, instead of sleep 10 on every restart. It also makes sure that not only is mysql running, but the DB credentials being used are working

```
zmdbuser=`cat /etc/zm/zm.conf | grep ZM_DB_USER | awk -F= '{print $2}' | tr -d ' '`
zmdbpass=`cat /etc/zm/zm.conf | grep ZM_DB_PASS | awk -F= '{print $2}' | tr -d ' '`
mysqlrunning=`/usr/bin/mysqladmin -u$zmdbuser -p$zmdbpass ping >/dev/null 2>&1 && echo RUNNING || echo NOTRUNNING`

echo  "Waiting for mysql to start..."
timeout=20
sleepval=4

while  [ "$mysqlrunning" != "RUNNING" ] && [ "$timeout" -gt "0" ]
do
	sleep $sleepval
	timeout=$((timeout-sleepval))
	mysqlrunning=`/usr/bin/mysqladmin -u$zmdbuser -p$zmdbpass ping >/dev/null 2>&1 && echo RUNNING || echo NOTRUNNING`
done

if [ "$mysqlrunning" == "RUNNING" ]; then
	retval=0
	echo "success"
else
	retval=-1
	echo "failed, mysql not started"
fi
```

Linwood-F, 2015-08-15:
Does the new systemd stuff deal with this automatically by its dependencies?

knnniggett, 2015-08-15:
Note that this problem will not exist on systems running systemd  ...dang it @Linwood-F you beat me to it.

pliablepixels, 2015-08-15:
Yup it won't. This is pre-systemd

pliablepixels, 2015-08-15:
pfft - so ubuntu 15 has moved to systemd it seems  - I guess its not worth it then

pliablepixels, 2015-08-15:
I'm going back to that lovely 2004 Rioja....

knnniggett, 2015-08-15:
Yes, the whole world has turned communist.
I, for one, welcome our new systemd overlords.

Seriously though, I keep finding good things with systemd ....like it has the built in ability to prevent zoneminder from starting if the event folder is not mounted (I just have no documented that yet).

closed 2015-08-24 by knnniggett