ZoneMinder/zoneminder #1024: feature-h264-videostorage, zm_videostore.cpp error when more than one client connected to RTSP stream

zylxpl, 2015-08-20:
I have bulid deb from feature-h264-videostorage, instal it and added ffmpeg camera. Stream is recorded using "H264 Camera Passthrough" which work in very efficient way. However, when I lunch RTSP stream in different app for live view something breaks down. In LOG i can find manny errors like:
Error encoding video frame packet: Invalid argument	zm_videostore.cpp and 'zmc -m 2' exited abnormally, exit status 255'. When i disconnect second stream everything goes back normal.
[image]

SteveGilvarry, 2015-08-20:
I will try and replicate this on the weekend, my first thought is the camera doesn't like two streams being delivered. Can you see if debug gives any more information on the frame being delivered or not from cam. At some point we will have to add a live view through zm.

knnniggett, 2015-08-20:
Along the lines of what @SteveGilvarry said, most IP cameras support dual streaming these days, but many require you to use different url's for each stream.  You need to verify how your particular make & model camera supports dual streams.  Look in the camera documentation. Look in the camera web page.  Does it talk about primary & secondary streams?  If so, look for details there.

Linwood-F, 2015-08-20:
My cameras allow a secondary stream, but they (and I thought most IP cameras) allow more than one connection to either stream.  The secondary stream is often used for a lower resolution feed that might suffice for motion detection while viewing or saving the high res feed, but mine can definitely have two systems access the same unicast stream number at the same time.

I mention this because the other thing I have found is that I experiment with multiple systems running zoneminder and other software with the same cameras, and I have found that they get overwhelmed.  Mine generally cannot run three streams reliably, and it appears to be some kind of a workload issue, though I have not dived into the details.  One thing you might try (though it may not be usable in the long run) is drop the frame rate substantially or the resolution and see if the problem goes away.  That might tell you if the problem relates to ability to handle both streams from a capacity standpoint or if it is more fundamental.

SteveGilvarry, 2015-08-20:
@zylxpl I had no issues with this when I used built-in web server on camera for live view, but this is plugin based so maybe that is the difference. What other app were you using for RTSP live view? But in saying that I know I have had two zm servers pulling the same RTSP in the past when I was developing.
Try turning debug on with zmc_m2 in extra component and 5 or 6 dbg level, should show packets as they arrive, mine just kept churning them through.

wdeninger, 2015-09-08:
Hi all,

I too have had cronic problems with av_interleaved_write_frame returning 'Invalid argument' (in zm_videostore.cpp at the tail end of VideoStore::writeVideoFramePacket(..) ).   I am running zoneminder using a passthrough mode with 3 HD cameras (1280x720@12fps, IFrames every 12th frame) on Mocord.  Errors generally occur between 10 minutes to1hr from zoneminder start and seem quite random.

The problem I have with this error (aside from not wanting to see it) is how to recover.  If the function 'Fatal' is called following the failure (as is currently found in the Git repo), the zmc process terminates and is restarted.  Unfortunately, restarting zmc also results in overwriting the passthrough time segment video.mp4 data (reuses filename).  For example, if  av_interleaved_write_frame failes 9 minutes into the time segment, the video.mp4 for the time segment will contain only minute (i.e 10:49 through 10:50).  

However, if the 'Fatal' call is modified to 'Warning',  the passthrough video.mp4 data is then found to be perserved and is complete -- with no errors at all.  But... the motion analysis frames become garbled and false motions events trigger continually.  I've implemented an ugly workaround to the problem by flagging the av_interleaved_write_frame failure and only calling 'Fatal' at the end of the time segment (in the VideoStorage destructor).  That way, I keep the integrity of the raw video data and only lose motion detect within the failed segment portion.

Do you believe this problem results from a bug in ffmpeg?  And is there a better way to recover from the av_interleaved_write_frame failure without terminating the passthrough stream?  Also... tangentially related, does there exist a method in 1.28.100 which allows separate rtsp streams for analysis and passthrough?  This would allow me to reduce motion analysis processing demands using a smaller resolution and frame rate,  yet maintain a high fidelity H264 passthough recording record.

-W
-W

wdeninger, 2015-10-28:
OK... I've found the problem with my case.  The libavformat function 

av_interleaved_write_frame(...)

call from zm_videostorage.cpp fails when repetitive calls to av_interleaved_write_frame are made with the same pts and dts values.  This DOES occur and is probably is not that unusual for high FPS streams (12 fps in my case) or low processing resource.  I believe the error resides within zoneminder.

As a 'temporary' workaround (atleast until the ZM gurus implement a proper fix) was to add a data member to track the previous pts and dts, and only allow the av_interleaved_write_frame to be called new image data was available.  The following are the changes to the zm_videostorage class:

src/zm_videostorage.h
32,34d33
>     int64_t prevPts;
>     int64_t prevDts;

src/zm_videostorage.cpp
118a119,120
>
>     prevPts = prevDts = 0;
178c181,186
<
---
>     if ( (prevPts!=opkt.pts) &&
>          (prevDts!=opkt.dts))  {
>     // deninger
>     prevPts = opkt.pts;
>     prevDts = opkt.dts;
>
184c193,196
<
---
>
>
>     };
>

-W

SteveGilvarry, 2015-10-30:
Thanks @wdeninger, inflex on irc has been having similar issues with out of sync DTS. I think Fatal is definitely the wrong error to throw at this point as it is a temporary hiccup. I was interested that you get corrupted images in the analysis stream as that is decoding the same packet. I would be interested to see what is happening using just ffmpeg to write the stream to disk, but would need a consistently failing camera to make it easy to identify when the issue occurs.

SteveGilvarry, 2015-10-30:
@wdeninger would be interested if recent pull https://github.com/ZoneMinder/ZoneMinder/pull/1130  works with your duplicates

wdeninger, 2015-10-31:
I performed an ffmpeg test using the CLI as shown below, and it DID result in the same permanently smudged/smeared state as does ZM after approximately 2 hours indicating this is an ffmpeg problem (image below). The stream was 1024x768 at 12fps storing images at 1fps using ffmpeg 2.7.2.

ffmpeg -i rtsp://username:password@ipaddress/mpeg4 -vf fps=1 img%05d.jpg

[image]
 
I will reproduce the above tonight using the most recent ffmpeg 2.8.1, but this time I'll remember to store the debug output log.

-W

wdeninger, 2015-10-31:
A continuation of the smudging/smearing problem above....

A repeat of the ffmpeg test log identified the following errors when the image stream becomes permanently smudged/smeared. 
.
.
[h264 @ 0x37baaa0] error while decoding MB 0 0, bytestream 166674
[h264 @ 0x37baaa0] concealing 3600 DC, 3600 AC, 3600 MV errors in I frame
[h264 @ 0x386a8c0] concealing 3368 DC, 3368 AC, 3368 MV errors in P frame
[h264 @ 0x38b1e40] concealing 3247 DC, 3247 AC, 3247 MV errors in P frame
[h264 @ 0x36d8880] concealing 3482 DC, 3482 AC, 3482 MV errors in P frame
[h264 @ 0x369eb60] top block unavailable for requested intra4x4 mode -1 at 5 0
[h264 @ 0x369eb60] error while decoding MB 5 0, bytestream 17855
[h264 @ 0x369eb60] concealing 3600 DC, 3600 AC, 3600 MV errors in P frame
[h264 @ 0x35efec0] top block unavailable for requested intra4x4 mode -1 at 49 0
[h264 @ 0x35efec0] error while decoding MB 49 0, bytestream 15051
[h264 @ 0x35efec0] concealing 3600 DC, 3600 AC, 3600 MV errors in P frame
[h264 @ 0x362aa00] concealing 3596 DC, 3596 AC, 3596 MV errors in P frame
frame= 4022 fps=1.0 q=24.8 size=N/A time=01:07:02.00 bitrate=N/A    ^M[h264 @ 0x372bbe0] concealing 3573 DC, 3573 AC, 3573 MV errors in P frame
.
.
An internet search revealed a thread of someone with similar problems ( http://ffmpeg.org/pipermail/ffmpeg-user/2014-July/022748.html ).  But unfortunately I had already attempted these unexplained remedies without having success:  I had updated ffmpeg to release 2.8.1, and configured/make/make installed it using permutations of all the options below:

--disable-asm --disable-inline-asm  --disable-optimizations

yet, no Joy..

-W

wdeninger, 2015-10-31:
Hmmm... more internet searching I do, the crazier it gets.  This thread ( https://trac.ffmpeg.org/ticket/1464 ) explores it possibly being a build chain issue.  For full disclosure, I'm using gcc 4.8.4, which should have solved the problem... and:

ffmpeg version 2.8.1 Copyright (c) 2000-2015 the FFmpeg developers
  built with gcc 4.8 (Ubuntu 4.8.4-2ubuntu1~14.04)
  configuration: --disable-inline-asm
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101

The problem is that ffmpeg coders have no idea.... I will take a careful look at all the ffmpeg compile warnings next.

wdeninger, 2015-10-31:
Oh... BTW, pull #1130 doesn't seem to generate any *-video.mp4 files anymore.  So there's some other problem with the build...  The monitor gives me the image, and analysis frames are generated, but no mp4 files are dropped when pass-through mode is selected.

-W

wdeninger, 2015-11-10:
The most recent 1.28.108 build I'm using (commit cb19113) seems to work wonderfully.  I especially love the 'analysis fps' feature.  Thanks guys!

** SOLVED **

As per the image smearing, I had discovered that the unrecoverable image smearing ONLY occurred when a "video encoding advanced parameters" setting for "private data" was enabled. This happened to be a default setting for the camera.  Once disabled, the problem vanished. This applied to both zoneminder and ffmpeg command line tests.

[image]

Sadly, this problem had been a thorn in my side for over 6 months. As such, I recommend anyone else having this problem perform a careful inspection or their camera options both documented and undocumented.

-W

knnniggett, 2015-11-10:
@wdeninger Way to hang in there. Programming instructions for your particular make/model of camera, especially your discovery about the Private data feature, would be good material for the wiki.

closed 2016-03-30 by SteveGilvarry