ZoneMinder/zoneminder #1025: Things to do to make API work out of the box for new PPAs

pliablepixels, 2015-08-20:
For @connortechnology to make it easy for the next build. Please feel free to close it when you merge them

1. Change directory permissions inside /use/share/zoneminder/www --> @connortechnology thinks it is one specific API temp diretory

2. Edit /etc/apache2/conf-enabled/zoneminder.conf and add this so .htaccess works
```
<Directory /usr/share/zoneminder/www/api>
    AllowOverride All
</Directory>
```
To get API security working, you also need to cherry pick this specific PR:
https://github.com/ZoneMinder/ZoneMinder/commit/ba3eaf6457c535d8171aafc08a739e0a89dac2ba

connortechnology, 2015-08-20:
The first is already fixed in my *_cmake package build scripts for debian based distros.  We use a link for /usr/share/zoneminder/www/api/app/tmp to /tmp/zm.
The second has already been applied to the package apache.conf file. 

I will take a look at the third later today.

packaging should happen faster from now on as I built a nice script to do it all for me.

pliablepixels, 2015-08-27:
Looks like:
a) The ZM_SESSION change in core.php
b) The file perm issue of APIs

are still not in master. I checked out today and built it.

I'm reopening this

pliablepixels, 2015-08-28:
closing. i see it now

closed 2015-08-28 by pliablepixels

knnniggett, 2015-08-29:
I'm wondering if it wouldn't be better to set the tmp folder inside core.php, rather than require each package maintainer to add a symlink for their zoneminder package.