ZoneMinder/zoneminder #1030: zmsystemctl.pl introduction in zmpkg.pl has broken isActive

pliablepixels, 2015-08-22:
It looks like zmpkg has been modified to call zmsytemctl which in turn seems to be calling zmpkg restart --> this breaks the isActive 1 setting as it is considered to be a new restart which clears out the bit. I'll dive in next week to study the new flow so I can make changes. For now, isActive will never show up

knnniggett, 2015-08-22:
Yes, due to nature of systemd, zoneminder has to do this. Please search this forum for past arguments of others who have not quite understood this.

knnniggett, 2015-08-22:
Instead of clearing the active state on a restart/reboot, if we instead always returned to the former state, wouldn't that get around this issue?

pliablepixels, 2015-08-22:
The issue that arises is that "running" (without a custom state) is not really a state defined in States table. So after setting a custom state there will be no way to remove the custom state.

knnniggett, 2015-08-22:
Right, but we can give all states a name in the States table if we want to change that.
Add an entry called "default", which puts the state back to default.

pliablepixels, 2015-08-22:
Does this sound reasonable?

zmpkg start --> start ZM with current isActive state
zmpkg restart -> start ZM with current isActive state
zmpkg default --> start ZM with default state (this is now a hard state in State DB)
zmpkg <statename> --> changes to <statename>

Modification to State.php:
a) In addition to listing custom states + start + stop + restart, we also show default (This should automatically happen as it picks all states from the DB wbich now has default)

b) We make sure the user cannot delete or rename default or create a new state with 'default'

knnniggett, 2015-08-23:
Sounds right.
Thinking out loud...
Should make it fail safe... if the system starts up and no state is active, then make default the active state.  ...or maybe we should call the default state "all"?

Not sure how easy it will be to prevent deleting the default state.  Alternatively, we could just re-add the state automatically if if is missing.

pliablepixels, 2015-08-25:
I've made the following changes. Will test for a few days and then do  a PR.
Seems to be working fine for me so far:

a) zm_update will create default in States if it does not exist
b) zm_create will create default in States
c) state.js has been modified so that you cannot create a new state that matches anycase "default"
d) state.js has been modified so that you cannot delete "default" (delete button is disabled if default)
e) zmpkg.pl.in has been modified such that start/restart does not mess which whatever state is active --> so state changes persist across reboots. 

I don't think I should be adding a check in zmpkg.pl.in to see if default has somehow been deleted - the only way that can happen is if someone directly access the DB - and in that case, other things in ZM can get messed up - so not sure if we need to implement a failsafe just for isActive - I can do it though - its easy enough, but will be a special case only for isActive when other items could also be deleted.

Can you think of any other entry points that I may have missed?

closed 2015-08-27 by pliablepixels