ZoneMinder/zoneminder #1042: server behind reverse proxy / ZM_BASE_URL

onlyjob, 2015-08-28:
As reported in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=770851

> I'm running ZM on standard port 80 internally but connecting through
a reverse proxy externally. The reverse proxy accepts connections on
a different port.
In this case zoneminder fails to display the video streams (monitor
window stays empty) due to hard coded URLs in the generated HTML.

> This has been reported in the ZM support forum here:

>  http://www.zoneminder.com/forums/viewtopic.php?f=31&t=22427

> which leads to the attached trivial patch. Please consider including this
patch in future releases.

~~~~
--- /usr/share/zoneminder/index.php.orig	2014-11-10 14:19:49.729657188 +0100
+++ /usr/share/zoneminder/index.php	2014-11-10 14:20:20.820586382 +0100
@@ -54,7 +54,7 @@ else
 {
     $protocol = 'http';
 }
-define( "ZM_BASE_URL", $protocol.'://'.$_SERVER['HTTP_HOST'] );
+define( "ZM_BASE_URL", "" );
 
 if ( isset($_GET['skin']) )
     $skin = $_GET['skin'];
~~~~

knnniggett, 2015-08-28:
So basically we need to change the URL's from absolute to relative.
I'm not sure why absolute url's were used in the first place.
That doesn't sound like it will break anything else, but I do want to test that first.

onlyjob, 2015-08-28:
I think idea of relative URLs makes sense but I did not have an opportunity to test this either...

mnbf9rca, 2015-12-27:
If you're using Apache you can use mod_proxy_html to fix this:

    <Location "/zm/">
       ProxyHTMLEnable On
       ProxyHTMLExtended On
       ProxyPass http://internalserver/zm/
       ProxyPassReverse /zm/
       RequestHeader    unset  Accept-Encoding
       SetOutputFilter  proxy-html
       ProxyHTMLURLMap http://internalserver/ /zm/ 
       ProxyHTMLURLMap  /zm/ /zm/
    </Location>
    <Location "/zm/cgi-bin/">
       ProxyHTMLInterp On
       ProxyHTMLEnable On
       ProxyHTMLExtended On
       ProxyPass http://internalserver/cgi-bin/
       ProxyPassReverse /
    </Location>

knnniggett, 2015-12-27:
This was previously fixed via cb7acb36ab08393d1fdcec84fe9d6754c88203ee
I simply forgot to close this issue.

closed 2015-12-27 by knnniggett

onlyjob, 2015-12-28:
Thanks. :smile: