ZoneMinder/zoneminder #1046: Optional google recaptcha

pliablepixels, 2015-08-29:
There was interest in adding a google recaptcha option to ZM which could be disabled/enabled via options. Default state is disabled. 

I'll be submitting a PR on this soon for review.

Note that the user will have to follow recaptcha instructions to configure a site and secret key for their ZM domain and enter those values in the options screen.

closed 2015-09-02 by pliablepixels

marcolino7, 2015-09-23:
Hi,
I updated my ZM installation from 1.28.102 to 1.28.106, and on login screen I got recaptcha widget with an error. I connot login on my installation.

[image]

marcolino7, 2015-09-23:
Hi,
I run database upgrade script, and now work. Sorry for this.
Marco