ZoneMinder/zoneminder #1051: zmupdate.pl sort on sql files incorrect

SteveGilvarry, 2015-09-06:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmupdate.pl.in#L1029
The sort here does not appear to get the order correct and runs 1.28.10x scripts before 1.28.99.
See here issue #935  https://github.com/ZoneMinder/ZoneMinder/issues/935#issuecomment-133233786
and #905  https://github.com/ZoneMinder/ZoneMinder/issues/905#issuecomment-138037678

knnniggett, 2015-09-06:
From a programming standpoint, the order in which the minor revision sql scripts does not matter.  The only dependency is the previous major revision.  I'm not saying this shouldn't be fixed, but I am saying that this is low priority.  It does not cause the update to fail. It merely looks strange.

The way to fix it is to perform a natural sort, rather than use the default filesystem sort order.

knnniggett, 2015-09-06:
@Linwood-F you want to look at this?

Linwood-F, 2015-09-06:
I only noticed it out of order, not that it did anything wrong, but I do have a question about your comment and how github and your merge process works.  Hypothetically, if a programmer issued .99 to fix something in SQL (say added field X as an int), then in .101 changed how that worked (say changed X to a float), might that not then conflict?  Or as part of merging to you require the correction be applied back against .99 in some fashion so that it doesn't produce two SQL updates against the same area of the schema? 

If the latter, then this is really a moot point and cosmetic.  If the former, I guess there's potential for harm.  

I'm clueless where the code lies that applies these in order, but I don't mind looking.  Whether it's worth doing or not I think depends on whether it can cause a problem.

knnniggett, 2015-09-06:
Yes, you are absolutely right. If the right number of issues align, then the order would matter and we would have a problem.  But this is something we'd know ahead of time.  Simply making sure we don't increase significant digits between the two hypothetical merges would avoid it.  Fixing the original merge w/o bumping the version number would work if the followup merge was made before we released a snapshot.

Again, I'm not saying it shouldn't be fixed.  It should be. I am saying that it can be avoided, and we certainly don't need to discuss it in other un-related threads.  That just makes solving those other issues harder.  Thank you @SteveGilvarry for creating this issue.  That was the right thing to do.

There are a lot of areas in ZoneMinder that need attention.  Just look at the list of open issues. Based on the circumstances, I'm not going to give this one a high priority.  Maybe someone else is willing to give it a shot.

@SteveGilvarry pointed out exactly where the issue is occurring.  The fix might be as simple as replacing Perl's sort command with a equivalent natural sort command. I don't know if there is such a thing in Perl, like there is in PHP.

Linwood-F, 2015-09-06:
When I get some time I'll take a look.  I went off on a tangent to a zabbix alternative (opennms).  My biggest issue in testing this is figuring out github enough to be able to pull down a specific version and then a later version so it has to run the update (i.e. a real, honest test instead of faked).  I think the key though has been answered -- while there is a possible failure mode, it's being managed, so it isn't going to happen for THIS wrap around.

knnniggett, 2015-09-07:
I've added an article to the Wiki, with the goal of helping others get through the github learning curve:
https://github.com/ZoneMinder/ZoneMinder/wiki/Understanding-Github-and-Pull-Requests

@pliablepixels could benefit from this as well.

pliablepixels, 2015-09-10:
I took a look at that code - I really don't think this is a bug in that line - its logic is correct. If we have moved from 2 digits to 3 digits in minor numbering, then we should change ".99" to ".999" and that would solve the problem correctly without having to change the code (which is doing it the right way)

SteveGilvarry, 2015-09-10:
.999 is still going to apply after .100, which is the problem, that being the sort is only looking at the next character and 1 is before 9 .99 was created first and in order of merge should be applying before all other sql scripts, in this case not an actual issue, but doesn't mean we don't need to find a better way.
http://search.cpan.org/~edavis/Sort-Versions-1.5/Versions.pm
Root cause analysis would lead me back to why we got a DB update called 1.28.99 as the first DB change, which leads me to the hard link between database version and application version.

pliablepixels, 2015-09-10:
I'm confused - I thought we want 10x to run before 99x? At least that is what your first post seemed to say is not happening?

pliablepixels, 2015-09-10:
Ah okay
So it seems for every 1.XXX.YYYY you want 1.XXX.99 to run first and then the rest. Is that  correct? I assumed the opposite.

SteveGilvarry, 2015-09-10:
Yes you added an x onto 99 above. Correct order is
1.28.99
1.28.100
1.28.101
Original post was "The sort here **does not** appear to get the order correct and runs 1.28.10x scripts before 1.28.99."
So we see the run order below
1.28.100
1.28.101
1.28.99
You were on the right track with setting the digits on the point release, which we don't, so we have 1.28.1 as a release, DB files as 1.28.99 and now 1.28.100. If you had said 1.28.99 should have been 1.28.099 that would have fixed current code. That perl module I linked handles the way we version but I am not a perl person so no idea if that module can be used.

knnniggett, 2015-09-10:
Guys,
While I agree that we could band-aid the issue with a very script numbering convention, the problem will always still be there because of the way Perl sort command works.

The desired behaviour @SteveGilvarry just mentioned is called a "natural sort".  PHP has builtin functions that handle this. So does Libreoffice Calc.  Maybe Perl does too.

SteveGilvarry, 2015-09-10:
I might have a go of this over the weekend, but here is what I am thinking. But first to admit that I know nothing about Perl.
`use Sort::Versions;`
`@files = sort versioncmp grep { (!/^\./) && /^zm_update\-[\d\.]+\.sql$/ && -f "$updateDir/$_" } readdir($dh);`

knnniggett, 2015-09-10:
Looks like Sort::Naturally is in both Ubuntu and CentOS core repos.  

Might be worth looking into that:
http://search.cpan.org/~bingos/Sort-Naturally-1.03/lib/Sort/Naturally.pm

pliablepixels, 2015-09-10:
given that perl allows you to specify the kind of operator compare, you could also do it without libraries, especially if naturally is not a core package

pliablepixels, 2015-09-10:
It seems perl has its own version library (mostly for parsing its own versions) starting 5.10. Is that old enough?

I wrote this modification - seems to print what you want. You might want to try running it and checking if the output is ok, incase I misunderstood the order again. Place it in share/zoneminder (outside of db folder)

```
use version;
$updateDir = "db";
opendir( my $dh, $updateDir ) || die "Can't open updateDir $!";
@files = sort { my ($x) = ($a =~ m/^zm_update\-(.*)\.sql$/);
                my ($y) = ($b =~ m/^zm_update\-(.*)\.sql$/);
                version->parse($x)  <=> version->parse($y)
                } grep { (!/^\./) && /^zm_update\-[\d\.]+\.sql$/ && -f "$updateDir/$_" } readdir($dh);
closedir $dh;
foreach my $patch ( @files ) {
        print $patch,"\n";
```

pliablepixels, 2015-10-04:
The current zmupdate.pl actually is buggy in a harmful way. In its currrent form, it can fail to apply patches and still mark your DB as upgraded.

While the line pointed to in this issue is benign in that it incorrectly sorts the order of files, the line that seems to cause harm  is https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmupdate.pl.in#L1039 

This error surfaced when I had installed @connortechnology PPA 1.28.2 which is what his stable branch is pushing out for the last couple of days. While 1.28.2 might have been pushed in error, it brought out what I think is a bug that causes harm while upgrading. When I ran ``zmupdate.pl``, the line referenced above (L1039) actually skipped all DB updates.

Here is the erroneous output: https://gist.github.com/pliablepixels/ff0d4cb270b978ca267b

I'll submit a PR to both correct the sort (which is benign) and the version compare (which is harmful)

Please review it works for you and using Perl versions is ok

closed 2015-10-06 by connortechnology