ZoneMinder/zoneminder #1053: Send images as inline attachment to display in the body of the mail

thefrip, 2015-09-09:
Hello,

I have modified zmfilter.pl so that images are considered as inline attachment that can be directly displayed in the body of the mail by many mail clients, instead of being treated as a mere attachment. Example of corresponding EMAIL_BODY:

<html><body>
[image]<br>
<br>
[image]<br>
<br>
Event Id : %EI%<br>
Length   : %EL%s<br>
Frames   : %EF% (%EFA%)<br>
Scores   : Tot:%EST% Max:%ESM% Avg:%ESA%<br><br>

<a href=%EPIM%>%EPIM%</a><br>
<a href=%EPS%>%EPS%</a><br><br>

%EI1% %EIM%
</body></html>

I have also added a test to make sure there is something to attach as we can run in deadlock condition otherwise when zoneminder cannot find the attachment.

I don't fork the git just for this and prefer to post my mod as a diff (based on latest version 1.28):

diff zmfilter.pl zmfilter.pl.org
1000,1009c1000,1005
<                 if (-e $attachment->{path}) {
<                   Info( "Attaching '$attachment->{path}'\n" );
<                   $mail->attach(
<                       Path => $attachment->{path},
<                       Type => $attachment->{type},
<                       Disposition => "attachment"
<                       );
<               } else {
<                   Info( "Cannot attach '$attachment->{path}'\n" );
<               }
---
>                 Info( "Attaching '$attachment->{path}\n" );
>                 $mail->attach(
>                     Path => $attachment->{path},
>                     Type => $attachment->{type},
>                     Disposition => "attachment"
>                 );
1021,1024d1016
<               Type => "multipart/related"
<           );
<
<           $mail->attach(
1029,1030d1020
<           my $attachment_count = 1;
<
1033,1044c1023,1028
<               if (-e $attachment->{path}) {
<                   Info( "Attaching '$attachment->{path}'\n" );
<                   $mail->attach(
<                       Path => $attachment->{path},
<                       Type => $attachment->{type},
<                       Encoding => "base64",
<                       Id => "attachment_$attachment_count"
<                   );
<                   $attachment_count++;
<               } else {
<                   Info( "Cannot attach '$attachment->{path}'\n" );
<               }
---
>                 Info( "Attaching '$attachment->{path}\n" );
>                 $mail->attach(
>                     Path => $attachment->{path},
>                     Type => $attachment->{type},
>                     Encoding => "base64"
>                 );

Hope this will be useful.
Rgds,

L.

knnniggett, 2015-09-10:
This is not a bad idea but if you want this added, you will need to generate a pull request.
Once that is complete, we will probably want to add a toggle under Options to allow the end user to select the type of behaviour.  I can step you through doing that if you make it that far.

thefrip, 2015-09-13:
Do I need to fork the git for that or can I do it in a "simpler way"? 
Just asking because I have a slow connection and I would like to avoid 
having to fork the entire git.

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

knnniggett, 2015-09-13:
If you want your feature added, you either need to be willing to do the work yourself or add a bounty to your request to pay someone else to do the work.

Our posting rules are documented here:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Here is how to use github:
https://github.com/ZoneMinder/ZoneMinder/wiki/Understanding-Github-and-Pull-Requests

Good luck!

knnniggett, 2015-09-28:
closing due to inactivity

closed 2015-09-28 by knnniggett