ZoneMinder/zoneminder #1059: Many ZoneMinder browser windows don't scale to content correctly

davenz, 2015-09-19:
These are seemingly minor usability issues which cumulatively end up consuming quite a bit of time. The client system is running Ubuntu 14.04, Firefox v40, all at 100 percent zoom. Server is ZoneMinder v1.28.1.

For example, the window to enable/disable a monitor (note the scroll bars because it's not scaling to content correctly):
[image]

The same for monitor settings:
[image]

And when opening a window to define zones:
[image]

Here is the same above window but maximized. This is the view I want to see initially, without having to manually scroll and maximize:

[image]

connortechnology, 2015-10-01:
Lots of these are fixed in master.

alager, 2015-10-02:
Configuring monitors is still not sized correctly in 1.28.107
[image]

commit 91ad6aff (2017-03-20): Increase default window sizes for the flat theme. Fixes #1059

Used the computed height of <html> and rounded up the nearest multiple of 5.
commit badbf1c7 (2017-03-20): Merge pull request #1816 from mnoorenberghe/flat_window_sizes

Increase default window sizes for the flat theme. Fixes #1059
closed 2017-03-18 by SteveGilvarry