ZoneMinder/zoneminder #1060: Ubuntu server 14.04 ltd x32 - Install with compilation - Failur Start

ychoucha, 2015-09-19:
Hello All,

 so i want use zoneminder for a usb webcam connected on my serveur with ubuntu 14.0 lts x32 (it's a old computer ) 

so after a frech ubuntu install i using this how to ---> 

```aptitude install -y apache2 mysql-server php5 php5-mysql build-essential libmysqlclient-dev libssl-dev libbz2-dev libpcre3-dev libdbi-perl libarchive-zip-perl libdate-manip-perl libdevice-serialport-perl libmime-perl libpcre3 libwww-perl libdbd-mysql-perl libsys-mmap-perl yasm automake autoconf libjpeg8-dev libjpeg8 apache2-mpm-prefork libapache2-mod-php5 php5-cli libphp-serialization-perl libgnutls-dev libjpeg8-dev libavcodec-dev libavformat-dev libswscale-dev libavutil-dev libv4l-dev libtool ffmpeg libnetpbm10-dev libavdevice-dev libmime-lite-perl dh-autoreconf dpatch;

```

```
ls -1 ~/zoneminder*;
```

```
$>sudo -s
$>dpkg -i /home/ayoub/zoneminder_1.28.1+1-trusty-SNAPSHOT2015071501_i386.deb ; apt-get install -f;
$>dpkg-reconfigure zoneminder;

```

after that i m trying to start zoneminder : 

```
$> sudo service zoneminder start
Starting ZoneMinder: failure
$>
```

and on var log i have this : 

```
Sep 19 15:41:46 srv01 zmpkg[13370]: INF [Sanity checking States table...]
Sep 19 15:41:46 srv01 zmpkg[13370]: INF [Command: start]
```

and this is my zm.conf : 
```
$> sudo nano /etc/zm/zm.conf
```

```
# ==========================================================================
#
# ZoneMinder Base Configuration, $Date$, $Revision$
#
# ==========================================================================
#
# This file is generated by 'configure'. Care should be taken if manually
# editing this file as an changes may be overwritten by subsequent configuration
# or installations.
#

# Path to installed data directory, used mostly for finding DB upgrade scripts
ZM_PATH_DATA=/usr/share/zoneminder

# Path to ZoneMinder binaries
ZM_PATH_BIN=/usr/bin

# Path to ZoneMinder libraries (none at present, for future use)
ZM_PATH_LIB=/usr/lib/i386-linux-gnu

# Path to ZoneMinder configuration (this file only at present)
ZM_PATH_CONF=/etc/zm

# Path to ZoneMinder web files
ZM_PATH_WEB=/usr/share/zoneminder

# Path to ZoneMinder cgi files
ZM_PATH_CGI=/usr/lib/cgi-bin

# Username and group that web daemon (httpd/apache) runs as
ZM_WEB_USER=www-data
ZM_WEB_GROUP=www-data

# ZoneMinder database type: so far only mysql is supported
ZM_DB_TYPE=mysql

# ZoneMinder database hostname or ip address
ZM_DB_HOST=localhost

# ZoneMinder database name
ZM_DB_NAME=zm

# ZoneMinder database user
ZM_DB_USER=zmuser

# ZoneMinder database password
ZM_DB_PASS=zmpass

# Host of this machine
ZM_SERVER_HOST=192.168.50.40

```

what do u think, how found, who blocking the server start please ?

knnniggett, 2015-09-19:
This issue does not meet our posting requirements because it does not describe a bug, nor is it a feature request. 

If you need help building zoneminder yourself, rather than use one of the existing repositories out there, then you should post this seek help in the zoneminder user forum.

Please refer to our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-09-19 by knnniggett

ychoucha, 2015-09-19:
no zoneminder is installed, is just i can't lunch it, and no log can explain why it's appening, so it"s bug because it's not clearing explain why zoneminder don"'t want lunch

knnniggett, 2015-09-19:
You have not provided any information at all indicating this is not a failure to install correctly. 

ZoneMinder is not a simple application. There are many ways to go off track.  However, you have chosen to install ZoneMinder the hard way. You can certainly do that, but there are not enough of us on the development team to assist every person who can't install the application.

To get help, you need to take this to the user forum. There plenty of users who will help you get zoneminder installed and working correctly on your system.