ZoneMinder/zoneminder #1061: One computer serving cameras to main server?

DaAwesomeP, 2015-09-21:
I want to have a specific setup. My main ZoneMinder machine will be Debian it will have a few cameras plugged into it. I also have an Ubuntu machine and a Windows machine with cameras on them. Is it possible to have the Ubuntu and Windows machine serve streams to the main Debian machine? I don't want to have any UI or Web access to the cameras on the Ubuntu and Windows machines. I effectively need those machines to act as network cameras.

knnniggett, 2015-09-21:
This issue does not describe a bug, nor is it a feature request.

Please refer to out posting rules and then post this in our user forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-09-21 by knnniggett

DaAwesomeP, 2015-09-21:
Ok. Sorry about that. You should create a `CONTRIBUTING.md` file in the root of the repo. GitHub will then show a banner when creating new issues and pull requests. You can just link to the Wiki in it.

Forum post: [One computer serving cameras to main server?](http://www.zoneminder.com/forums/viewtopic.php?f=32&t=23473)

knnniggett, 2015-09-21:
Our github posting rules are linked under the section Contacting the Development Team in Readme.md, which is prominently displayed on our front page in large front.

I'll look into duplicating that in contributing.md