ZoneMinder/zoneminder #1071: Shared data not initialised by capture daemon

bitnumus, 2015-09-28:
Right, before the rants about this being a non-issue, i have read some issues and dupes such as:

https://github.com/ZoneMinder/ZoneMinder/issues/629

However, I'm 99% sure this just caused my monitor to miss an event.

I have two streams, subStream(modect) and mainStream(nodect), they are linked therefore. I received this ERR log from the mainStream ZMA process. I had a recorded event on the subStream monitor but not the mainStream monitor. During this period i received this error message multiple times.

I'm happy to help debug, but i've read some replies to the threads saying this is a non-issue, now maybe the missed event is totally unrelated but i would like to explore that if possible!

connortechnology, 2015-09-28:
The message means that zmc isn't running.  The question is: why is zmc not running? Check your zmc logs.

Linwood-F, 2015-09-28:
Also see issue #984 which if you haven't included that code yet (it's merged but not in the prior release) that is one issue where zmc could be running but erroneously give this error.

knnniggett, 2015-09-28:
We've already patched master such that this message no longer appears.

So far you've not provided required information describing your system, how you've got it configured, and a detailed listing of your logs. 

This is not the place to have a discussion about why your system is not performing to your expectations.  The place for that is in our IRC channel or user forum. 

Please see our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Many people post here, thinking they've got a bug, only to find there system was misconfigured.  
Create a post in the user forum, with an detailed description of your system and logs, and then we will go from there.

closed 2015-09-28 by knnniggett