ZoneMinder/zoneminder #1077: ZoneMinder breaks after apt-get dist-upgrade

elvisimprsntr, 2015-09-29:
I have been successfully running 1.28.2 for a few weeks without issue.  Today I performed an

`apt-get update`
`apt-get dist-upgrade`

of my system.

#### ZoneMinder was upgrade to 1.28.106.  

`oot@antec-nvr:/home/alec# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... The following packages were automatically installed and are no longer required:
  ffmpeg libavdevice-ffmpeg56 libavfilter-ffmpeg5 libavresample-ffmpeg2
  libbs2b0 libexpect-perl libio-stty-perl libpgm-5.1-0 libpostproc-ffmpeg53
  libsodium13 libzmq3 php5 qt-faststart
Use 'apt-get autoremove' to remove them.
Done
The following NEW packages will be installed:
  libav-tools libavdevice55 libavfilter5 libimage-base-bundle-perl
  libimage-info-perl libjs-mootools zoneminder-doc
The following packages will be upgraded:
  zoneminder
1 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,157 kB of archives.
After this operation, 26.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libavdevice55 amd64 6:11.2-1 [33.1 kB]
Get:2 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu/ vivid/main zoneminder amd64 1.28.106-vivid-2015092901 [2,943 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libavfilter5 amd64 6:11.2-1 [113 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libav-tools amd64 6:11.2-1 [416 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libimage-base-bundle-perl all 1.0.7-3 [30.3 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libimage-info-perl all 1.28-1 [63.0 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu/ vivid/universe libjs-mootools all 1.4.5~debian1-2.1 [241 kB]
Get:8 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu/ vivid/main zoneminder-doc all 1.28.106-vivid-2015092901 [5,317 kB]
Fetched 9,157 kB in 5s (1,766 kB/s)         
Selecting previously unselected package libavdevice55:amd64.
(Reading database ... 186210 files and directories currently installed.)
Preparing to unpack .../libavdevice55_6%3a11.2-1_amd64.deb ...
Unpacking libavdevice55:amd64 (6:11.2-1) ...
Selecting previously unselected package libavfilter5:amd64.
Preparing to unpack .../libavfilter5_6%3a11.2-1_amd64.deb ...
Unpacking libavfilter5:amd64 (6:11.2-1) ...
Selecting previously unselected package libav-tools.
Preparing to unpack .../libav-tools_6%3a11.2-1_amd64.deb ...
Unpacking libav-tools (6:11.2-1) ...
Selecting previously unselected package libimage-base-bundle-perl.
Preparing to unpack .../libimage-base-bundle-perl_1.0.7-3_all.deb ...
Unpacking libimage-base-bundle-perl (1.0.7-3) ...
Selecting previously unselected package libimage-info-perl.
Preparing to unpack .../libimage-info-perl_1.28-1_all.deb ...
Unpacking libimage-info-perl (1.28-1) ...
Selecting previously unselected package libjs-mootools.
Preparing to unpack .../libjs-mootools_1.4.5~debian1-2.1_all.deb ...
Unpacking libjs-mootools (1.4.5~debian1-2.1) ...
Preparing to unpack .../zoneminder_1.28.106-vivid-2015092901_amd64.deb ...
Unpacking zoneminder (1.28.106-vivid-2015092901) over (1.28.1-vivid-1) ...
Selecting previously unselected package zoneminder-doc.
Preparing to unpack .../zoneminder-doc_1.28.106-vivid-2015092901_all.deb ...
Unpacking zoneminder-doc (1.28.106-vivid-2015092901) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for systemd (219-7ubuntu6) ...
Processing triggers for ureadahead (0.100.0-19) ...
ureadahead will be reprofiled on next reboot
Setting up libavdevice55:amd64 (6:11.2-1) ...
Setting up libavfilter5:amd64 (6:11.2-1) ...
Setting up libav-tools (6:11.2-1) ...
Setting up libimage-base-bundle-perl (1.0.7-3) ...
Setting up libimage-info-perl (1.28-1) ...
Setting up libjs-mootools (1.4.5~debian1-2.1) ...
Setting up zoneminder (1.28.106-vivid-2015092901) ...
Installing new version of config file /etc/zm/zm.conf ...
Setting up zoneminder-doc (1.28.106-vivid-2015092901) ...
Processing triggers for libc-bin (2.21-0ubuntu4) ...
Processing triggers for systemd (219-7ubuntu6) ...
Processing triggers for ureadahead (0.100.0-19) ...`

#### Now receive the follow errors when checking status.  

`root@antec-nvr:/home/alec# systemctl status zoneminder.service
● zoneminder.service - ZoneMinder CCTV recording and surveillance system
   Loaded: loaded (/lib/systemd/system/zoneminder.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2015-09-29 18:57:58 EDT; 7min ago
  Process: 1363 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=255)

Sep 29 18:57:56 antec-nvr systemd[1]: Starting ZoneMinder CCTV recording and surveillance system...
Sep 29 18:57:57 antec-nvr zmpkg[1363]: INF [Sanity checking States table...]
Sep 29 18:57:57 antec-nvr zmpkg[1363]: INF [Fixing States table - either no default state or duplicate defa...ates]
Sep 29 18:57:58 antec-nvr zmpkg.pl[1363]: DBD::mysql::st execute failed: Unknown column 'IsActive' in 'fiel... 333.
Sep 29 18:57:58 antec-nvr zmpkg[1363]: FAT [Can't execute: Unknown column 'IsActive' in 'field list']
Sep 29 18:57:58 antec-nvr systemd[1]: zoneminder.service: control process exited, code=exited status=255
Sep 29 18:57:58 antec-nvr systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
Sep 29 18:57:58 antec-nvr systemd[1]: Unit zoneminder.service entered failed state.
Sep 29 18:57:58 antec-nvr systemd[1]: zoneminder.service failed.
Hint: Some lines were ellipsized, use -l to show in full.`    

#### Originally installed following these instructions:     

http://www.zoneminder.com/wiki/index.php/Ubuntu_Server_15.04_64-bit_with_Zoneminder_1.28.1_the_easy_way

pliablepixels, 2015-09-29:
I think you  forgot to update the Database by ```sudo zmupdate.pl```

elvisimprsntr, 2015-09-29:
OK, now get the follow messages.    

 As root: (after confirming ZM was not running)

`zmupdate.pl`

`root@antec-nvr:/home/alec# zmupdate.pl  

Initiating database upgrade to version 1.28.106 from version 1.28.2

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop : 

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.28.106
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.99 from 1.28.2
Warning: Using a password on the command line interface can be insecure.
ERROR 1142 (42000) at line 393: CREATE command denied to user 'zmuser'@'localhost' for table 'Servers'
Output: 
Command 'mysql -hlocalhost -uzmuser -pzmpass zm < /usr/share/zoneminder/db/zm_update-1.28.99.sql' exited with status: 1`

pliablepixels, 2015-09-29:
That's the other "breaking change" - please take a look at the ZM forums - I think either @knnniggett  or @connortechnology posted that you now need to grant permissions to the DB, if I am not mistaken

pliablepixels, 2015-09-30:
Here you go - http://www.zoneminder.com/forums/viewtopic.php?f=32&t=23375&p=89206&hilit=grant#p89206 - you'll find the grant command inside the set of instructions bbunge posted

knnniggett, 2015-09-30:
This does not describe a bug but a failure to upgrade properly.
Please seek further help in the zoneminder user forum.

closed 2015-09-30 by knnniggett

connortechnology, 2015-09-30:
Actually what happened is that I accidentally pushed a master package to the real releases ppa.  It was only there for a minute, but looks like you managed to do your update in that time.

You might try apt-get remove zoneminder 
apt-get update
apt-get install zoneminder

Which should take you back to 1.28.2

connortechnology, 2015-09-30:
ok, back to 1.28.1.  .2 has never been released.

elvisimprsntr, 2015-09-30:
@connortechnology 

keeps installing 1.28.106

pliablepixels, 2015-09-30:
I think he is on your master branch

elvisimprsntr, 2015-09-30:
####  From wiki instructions:  http://www.zoneminder.com/wiki/index.php/Ubuntu_Server_15.04_64-bit_with_Zoneminder_1.28.1_the_easy_way  

As of 03JUL15 there is a package in the PPA for Vivid Zoneminder. This how to has been edited to reflect the change.
If you install Ubuntu from server CD, install OpenSSH Server, and LAMP Server (LAMP installs Apache2, 

PHP, and MySQL server)
If you use the minimum Ubuntu ISO CD, install Basic Server, OpenSSH Server, and LAMP Server (LAMP installs Apache2, PHP, and MySQL server)
Note: This install is for the iconnor/zoneminder PPA. As of 19AUG15 this installs Zoneminder 1.28.2.

#### 106 still available:  https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master

SteveGilvarry, 2015-09-30:
You missed the next key line
*There is a newer version of Zoneminder in the iconnor/zoneminder-master PPA but my test install did not include an automatic database install and I could not get this version of Zoneminder to work.*
The instructions in the wiki are correct
add-apt-repository ppa:iconnor/zoneminder, which is https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder
Looks like you did
add-apt-repository ppa:iconnor/zoneminder-master
Which is where .106 is.

connortechnology, 2015-09-30:
ah yes.  I was just assuming my screwup today was the problem.

elvisimprsntr, 2015-09-30:
add-apt-repository -r ppa:iconnor/zoneminder-master
add-apt-repository ppa:iconnor/zoneminder
apt-get purge zoneminder
apt-get update
apt-get install zoneminder

still install 106

elvisimprsntr, 2015-09-30:
root@antec-nvr:apt-cache policy zoneminder
zoneminder:
  Installed: (none)
  Candidate: 1.28.106-vivid-2015092901
  Version table:
     1.28.106-vivid-2015092901 0
        500 http://ppa.launchpad.net/iconnor/zoneminder/ubuntu/ vivid/main amd64 Packages
     1.26.5-1ubuntu7 0
        500 http://us.archive.ubuntu.com/ubuntu/ vivid/universe amd64 Packages

connortechnology, 2015-09-30:
Yeah.. that's what I thought, you are on the release ppa, and got the mistake... 
Thing is, I have removed that from the ppa... but you are still getting it.  
Maybe I'll have to do a very small .2 release just to fix this disaster.

elvisimprsntr, 2015-09-30:
Anything to keep someone else from getting bit.  

I think my DB was already at 1.28.2 when I encountered this problem today, but you indicted 1.28.1 is what it will (eventually) try to reinstall.  Will that be a problem?

connortechnology, 2015-09-30:
I'm not sure what I can do at this point.  The ppa is still giving out the crap version despite it being deleted 13 hours ago.  

I guess we will see what the status is in the morning.  

1.28.2 has never been released.  I feel that you are mistaken about your previous version #.

connortechnology, 2015-09-30:
So.. the ppa is still screwed up.  ubuntu PPA package deletion it clearly full of lies. 

I have deleted the entire PPA.  Hopefully this will prevent anyone else from getting the unwanted upgrade.   Then I will re-create the ppa.

Of course, even the ppa deletion is not happening quickly.

elvisimprsntr, 2015-10-03:
Looks like the PPA is at 1.28.1 again.   Thanks for fixing.   I tried to reinstall, but keep getting dependency  and other "missing" issues.  I ended up having to rebuild my ZM server from scratch and set everything up again.