ZoneMinder/zoneminder #1081: Nodect goes into alarm state without any external trigger

mdetweiler, 2015-09-30:
If you set a monitor to Nodect it still detects motion and creates events.  I believe this should only happen through an external trigger.

knnniggett, 2015-10-01:
I'm not seeing this behaviour. 

I'm currently monitoring one of Foscam's demo camera's in ZoneMinder 1.28.1 using "nodect".  Many people are walking in front of the camera, yet the camera's status remains at idle and no events are generated.

You should verify you aren't unknowingly generating external events.  Verify your camera is not linked to another.

1.27.99 was a development release. If you are still experiencing an issue, you should upgrade to the most recent release of zoneminder. Following that, you should describe your system and camera configuration in the user forum, as this does not appear to be a bug.

closed 2015-10-01 by knnniggett

Linwood-F, 2015-10-01:
Also, if I recall despite the documentation, nodect will respond to linked events from other monitors as well as external, if that is what you are seeing.

mdetweiler, 2015-10-01:
So the interesting thing that I am doing is updating the Time stamp label with temperature information using an old trick with zmfilter.pl, but that only happens every 15 minutes and events are generated constantly when there is motion in front of any of the cameras.  Setting the monitor to Nodect is required for the trick to work, but otherwise I would just set them to "Monitor".  I tried quickly to edit zmfilter.pl so that monitors that were set to "Monitor" would be recognized, but I think there is no socket available unless they are set to one of the original 3 states.  

I use a bunch of different camera setups that are all basically IP cameras, but have never noticed this behaviour in past releases.  I have no linked monitors setup.  Perhaps an upgrade will resolve it.

mdetweiler, 2015-10-01:
Is there an easy way to upgrade this release on Ubuntu?  I've seen some instructions that suggest a few apt commands could do this, but I'm concerned that I could bring down the system permanently.

Thanks.

knnniggett, 2015-10-01:
@mdetweiler please refer to user forum and wiki for discussions not involving bug fixes or feature requests. There is some excellent "easy-way" documentation in the wiki for Ubuntu users.

mdetweiler, 2015-10-01:
@knnniggett Asking for a pointer or advice on upgrading from what you just described as a development release seems on topic to me.  Especially since you suggested an upgrade might resolve the issue that still seems like a bug to me.  Relax...  

A more helpful reply might have been to describe in detail what Nodect is really doing under the covers and why you think it could possibly ever go into alarm state without any external trigger or if zmfilter interacts with the socket in a different way when Nodect is used.

I've supported ZoneMinder for many years, financially, contributing code and to the old forums.  I appreciate the community that has kept it alive for so long, especially those that use a more helpful and thankful tone.

connortechnology, 2015-10-01:
The information is a quick google away.  What more could we do than google it ourselves and give you a link?

You must not be using a package or your system would have automatically been updated.
I suggest removing all traces of your previous install (except db content of course) and using my ppa https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder

Otherwise, build your system the same way you did before, but using a more recent source tree.

the db update is then done by running zmupdate.pl

knnniggett, 2015-10-01:
@mdetweiler When you created this issue, github encouraged you to read our guidelines for posting, which is documented here: https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules .

Our github forum is being reserved for bug reports and serious feature requests only. This has been implemented because the few of us who chose to volunteer our time on this project need a way to know what are issues that likely require changes to the source code and which are not.  Zoneminder has a very large user base and the ~3 of us cannot handle it all by ourselves.  Keeping non-bug related discussions in the user forum, helps us significantly.  You'll find that I spend a lot of time in the user forum myself, so posting there does not mean you won't get the same level of support.

If you would take the time to read the ZoneMinder WiKi, you would find that a user by the name of bbunge has spent a lot of time creating "easy-way" documentation that answers the exact upgrade question you just asked. I'm absolutely not going to repeat back to you what he has already documented.

We all here appreciate your support of this project, but that does not mean you are entitled to special treatment.  We've implemented a set of rules, which we expect everyone to follow. This helps the project flow with the limited number of volunteers that choose to maintain it.