ZoneMinder/zoneminder #1086: Failures during upgrade Ubuntu 14.04 ZM 1.27.99 to 1.28.1

mdetweiler, 2015-10-01:
Installed using conner ppa

So...  after attempting an upgrade as you suggested to resolve the bug #1081 I was seeing, I ran into bug #935 .  I was sort of looking for a hint like that when I asked about the upgrade, but anyway...  I also hit an issue with my apache links becoming invalid and the webroot seeming to change, but the biggest problem I have now is that I am not getting any video at all on any monitor.  I can't find anything addressing this in the searches on upgrades on Ubuntu 14.04 I've done so far so if it isn't too much trouble to ask...  any idea what might have caused this?  My guess is something to do with ffmpeg since I had to add a special ppa for that when I did the original install, but I don't recall why and it was not updated during the apt-get update process (it failed to be found). 

P.S.  The wiki is great, but doesn't talk about upgrade problems from development releases...

connortechnology, 2015-10-01:
Yeah, it hasn't updated your db.  You need to run zmupdate.pl.  most of the process won't run if the db is't up to date.

mdetweiler, 2015-10-01:
I did run that and ran into the permissions issue just like the bug thread and ran it again and it claimed to succeed.

mdetweiler, 2015-10-01:
Just tried to run again and:

root@system-z:~# service zoneminder stop
Stopping ZoneMinder: success

root@system-z:~# zmupdate.pl

Database already at version 1.28.107, update aborted.

connortechnology, 2015-10-01:
ah, damn that permissions problem. 

echo 'grant lock tables, create, alter on zm.* to 'zmuser'@localhost identified by "zmpass";' | mysql --defaults-file=/etc/mysql/debian.cnf mysql

mdetweiler, 2015-10-01:
No change...  Interestingly, the cameras seem to be capturing and events are generating so it's just the UI?  Perhaps something to do with the change in the apache configs?

connortechnology, 2015-10-01:
probably apache cgi-bin config. the cgi-bin path changed from /cgi-bin to /zm/cgi-bin and the actual path chagned from /usr/lib/cgi-bin to /usr/lib/zoneminder/cgi-bin

mdetweiler, 2015-10-01:
Yea..  that's the issue.  I can access the streams in FireFox if I give the correct url.

mdetweiler, 2015-10-01:
I'm confused about which config files are needed and where they should be.

These were some old instructions I think...
ln -s /etc/zm/apache.conf /etc/apache2/conf-available/zoneminder.conf
ln -s /etc/apache2/conf-available/zoneminder.conf /etc/apache2/conf-enabled/zoneminder.conf

mdetweiler, 2015-10-01:
I'm having no luck fixing it.  I have restored all of my other apache config files to their original since I was running other sites on this server and still can't get images in the UI.   Any ideas?
Restarting apache of course...

mdetweiler, 2015-10-01:
The apache error log definitely shows a problem when I try to view a monitor:

[Thu Oct 01 16:25:47.330838 2015] [cgi:error] [pid 3476] [client 38.140.13.162:49283] script not found or unable to stat: /usr/lib/cgi-bin/nph-zms, referer: http://www.systemxxx/zm/index.php?view=watch&mid=1
[Thu Oct 01 16:25:49.026730 2015] [:error] [pid 3478] [client 38.140.13.162:49265] ERR [socket_sendto( /tmp/zm/zms-608039s.sock ) failed: No such file or directory], referer: http://www.systemxxx/www/index.php?view=watch&mid=1
[Thu Oct 01 16:25:50.010204 2015] [:error] [pid 2717] [client 38.140.13.162:49284] ERR [socket_sendto( /tmp/zm/zms-295181s.sock ) failed: No such file or directory], referer: http://www.systemxxx/zm/index.php?view=watch&mid=1
[Thu Oct 01 16:25:53.194058 2015] [:error] [pid 2717] [client 38.140.13.162:49284] ERR [socket_sendto( /tmp/zm/zms-295181s.sock ) failed: No such file or directory], referer: http://www.systemxxx/zm/index.php?view=watch&mid=1

Looks like the UI is still trying to point to the old cgi-bin directory?

connortechnology, 2015-10-01:
Go into your ZM options Paths tab.  It probably still has the cgi-bin path set to /cgi-bin/zms-nph. change that to /zm/cgi-binetc

mdetweiler, 2015-10-01:
Nice!  That fixed it!  Thanks for your help and patience today!  And also for the ppa that I've been using since you started it.

.....
Interestingly, I am still seeing monitors go into alarm state when they are set to Nodect, but the bug was closed as you know...  #1081 so I'll try to figure that out I guess...

knnniggett, 2015-10-02:
Closing.

To make progress on why your system is picking up external triggers in nodect, I will say again that your best option is to post into the user forum. Describe your system and how you've got it set up. 

By doing so, you will be casting a much wider net. There is likely someone out there who is doing nearly the same the thing you are and can help make inroads on this.  

At the moment, there is no data to indicate this is a bug, but if anything significant comes out of the discussion in the user forum, we will be glad to make changes on our end.

closed 2015-10-02 by knnniggett

mdetweiler, 2015-10-02:
I will attempt to reproduce it on a separate clean install and take it from there.

ignik, 2015-11-28:
ln -s /usr/lib/zoneminder/cgi-bin/nph-zms /usr/lib/cgi-bin/

tommycw1, 2015-11-29:
All, just to follow up on this, I had a similar issue.  I was running 1.28
on an older ubuntu install, then upgraded to 15.04, then 15.10 in the same
week.  This broke my ZM install and the issue was this CGI-BIN path.
Wonder if we can put something into the package to look for this issue and
fix on upgrade?

Tom

ignik, 2015-11-30:
1. goto https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master and 
cat /etc/apt/sources.list.d/iconnor-ubuntu-zoneminder-master-wily.list
deb http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu wily main 
deb-src http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu wily main 
2. update - upgrade - not working ;-)
3. `chmod a+r /etc/zm/zm.conf`
4. `service zoneminder start`, show `systemctl status zoneminder.service`
5. `a2enconf zoneminder` (and repeat 4)
6. ln -s /usr/lib/zoneminder/cgi-bin/nph-zms /usr/lib/cgi-bin/` (and repeat 4.)
7. if you have incorrect db, remove old and make new
echo "drop database zm;; quit;" | mysql  -uroot -p 
mysql -u root -p zm < /usr/share/zoneminder/db/zm_create.sql
and repeat 4.

Or, if you have very sexy /var/cache/zoneminder/events/  -
show it and too much peoples can help repair it ;)

connortechnology, 2015-11-30:
instead of #3, how about chgrp www-data /etc/zm/zm.conf.

knnniggett, 2015-11-30:
Better - read the "Easy Way" documentation in the user forum, which shows how to set up @connortechnology 's ppa and install zoneminder on Ubuntu. It avoids the configuration problems mentioned in this github issue.
https://wiki.zoneminder.com/Ubuntu