ZoneMinder/zoneminder #1091: Sound alert in montage view

segordon, 2015-10-03:
The option exists to enable sound alerts for individual zone live-view on alarm (WEB_ALARM_SOUND), however this feature does not extend into Montage view. 

As the montage view already automatically resizes whatever feed is currently alarming, it seems as if it would be trivial to add an option to allow for the WEB_ALARM_SOUND to fire in montage view, too.

In single zone live-feed mode, the sound is triggered by logic within /js/watch.js.

snippet of watch.js
```javascript
       if ( SOUND_ON_ALARM )
        {
            // Enable the alarm sound
            if ( !canPlayPauseAudio )
                $('alarmSound').removeClass( 'hidden' );
            else
                $('MediaPlayer').Play();
```

There is this in montage view..  /js/montage.js

snippet of montage.js
```javascript
            if ( newAlarm )
            {
                if ( false && SOUND_ON_ALARM )
                {
                    // Enable the alarm sound
                    $('alarmSound').removeClass( 'hidden' );
                }
                if ( POPUP_ON_ALARM )
                {
                    windowToFront();
                }
            }
            if ( false && SOUND_ON_ALARM )
            {
                if ( oldAlarm )
                {
                    // Disable alarm sound
                    $('alarmSound').addClass( 'hidden' );
                }
            }
        }
```
Can the difference between the statements including the if/else involving $('MediaPlayer').Play() be the difference in behavior? Was sound originally supposed to be functional for Montage view, as well?

mkauzlar, 2016-11-30:
I vote for this to be implemented (sound an alarm and focus on camera when in montage view)

kennbr34, 2018-04-07:
I have created a fix against the 1.30.4 that works on Chrome Version 65.0.3325.181

The files edited are:

js/montage.js
js/montage.js.php
montage.php

Here's the diff

[montagealarmsound.txt](https://github.com/ZoneMinder/zoneminder/files/1886775/montagealarmsound.txt)

closed 2018-06-22 by stale[bot]