ZoneMinder/zoneminder #1098: 1.28.2 zmfilter.pl syntax error

mdetweiler, 2015-10-08:
root@ubuntu-video:~/bin# /usr/bin/zmfilter.pl 
Bareword found where operator expected at /usr/bin/zmfilter.pl line 828, near "Error"
	(Missing semicolon on previous line?)
syntax error at /usr/bin/zmfilter.pl line 828, near "Error"
Execution of /usr/bin/zmfilter.pl aborted due to compilation errors.
root@ubuntu-video:~/bin# vi /usr/bin/zmfilter.pl

root@ubuntu-video:~/bin# /usr/bin/zmfilter.pl 
Name "Net::SFTP::Foreign::debug" used only once: possible typo at /usr/bin/zmfilter.pl line 818.

knnniggett, 2015-10-22:
I was hoping @connortechnology would respond to this.
Zoneminder 1.28.2 only exists in his Ubuntu ppa. I don't have a copy of that to see what you've got. 
The copy of zmfilter.pl in master is clearly different than the one you are using:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmfilter.pl.in

You can ignore the message about Net::SFTP::Foreign::debug. It is just a warning and can be ignored. I don't think the author of Net::SFTP::Foreign wrote a debug module, so we simply need to disable the check for debug when using sftp as opposed to plain ftp. We are aware of it and it will get fixed eventually.

connortechnology, 2015-10-22:
I think the issue is a missing ; after the eval above that line.  When doing block evals I think you need a semi-colon after the end of the block;

So on line 827 change it to 

}; # end eval

mdetweiler, 2015-10-22:
Yes...  I had fixed it that way when I saw this issue on my system.

knnniggett, 2015-10-22:
Can someone with 1.28.2 compare their copy of zmfilter.pl to the copy in master and verify whether or not this problem hasn't already been fixed? 

Line 907 looks suspect:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmfilter.pl.in#L907

connortechnology, 2015-10-22:
The issue does not exist in master.  It exists only in 1.28.2.  Although I am going to look at merging the code into master.  The Net::SFTP stuff uses die and croak, so it can kill zmfilter.pl.  So we need to put it into an eval to prevent the failure of one filter from killing all filters.

knnniggett, 2016-03-09:
@mdetweiler If you are still around, can you test #1323 ?
It makes the Net::SFTP::Foreign::debug warnign go away, but I want to make sure that the sftp transfers still work correctly.

mdetweiler, 2016-03-09:
Did you want me to test it on 1.29?

knnniggett, 2016-03-09:
You should be able to patch any recent version of zoneminder with the proposed change.

knnniggett, 2016-03-21:
The original syntax error from this issue was fixed previously.
#1323 fixes the warning that was also mentioned in this issue.

closed 2016-03-21 by knnniggett

Mat2Vence, 2016-04-28:
I am having the same kind of issue with the 1.29.

mdetweiler, 2016-04-29:
I just did an install of 1.29 from the Ubuntu 14.04 PPA and did not see this issue.  Are you using Ubuntu or another OS?

Mat2Vence, 2016-04-29:
I am running on Ubuntu 14.04.4 LTS