ZoneMinder/zoneminder #1104: PHP Warnings

baffo32, 2015-10-12:
After a fresh install of zoneminder on my gentoo system, the console shows mostly PHP warnings.

[image]

knnniggett, 2015-10-12:
You need to do what it says and set date.timezone in your php.ini
This is a standard requirement for most distros.

baffo32, 2015-10-12:
Thanks, now all I have left is
> Warning: disk_total_space(): No such file or directory in /usr/share/zoneminder/www/includes/functions.php on line 1642
> Warning: disk_free_space(): No such file or directory in /usr/share/zoneminder/www/includes/functions.php on line 1643
> Warning: Division by zero in /usr/share/zoneminder/www/includes/functions.php on line 1643

This appears to be because ZM_DIR_EVENTS is unset.

baffo32, 2015-10-12:
Ah-ha, I'm missing the events folder

knnniggett, 2015-10-12:
Installation instructions for Gentoo are here:
https://wiki.gentoo.org/wiki/ZoneMinder

Please review it as the instructions for editing php.ini, along with many other required configuration steps, are there.

You appear to be attempting a from source installation, in which case you should also follow these instructions here: 
https://github.com/ZoneMinder/ZoneMinder/blob/master/INSTALL

This documents the use of zmlinkcontent.sh among other important steps.

ZoneMinder is not easy to build and install from source even for advanced users, which is why we recommend you install from package instead.  

If you need additional support then please move this conversation to the user forum or our IRC channel as our github forum is reserved for bug reports and feature requests only.
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-10-12 by knnniggett