ZoneMinder/zoneminder #1115: Release v1.29.0 (take two)

knnniggett, 2015-10-22:
Supersedes #549

This is my attempt to document what we need to complete before we release:

@knnniggett
- zms is causing memory corruption in some of my test environments. I've spent quite a bit of time on this. I identified the line of code causing the crash but have so far not been able to determine why. Will create an issue and document what I've done and get others to look at it.

- Will scan the commits since 1.28.1 release and create a "what's new" list.  I expect to come across new features that still need to be documented.

- Remove absolute URL's as identified in #1042. Trivial.

@connortechnology 
- Finish multiserver #1040 
- Create initial documentation and/or help text for multiserver

@pliablepixels 
- Confirm we've reached a good stopping point for development of the API
- Keep all his cool new ideas to himself until after we release, so our heads don't explode. :-)

@kylejohnson 
Fix #1085 . Sounds like we just need to add an index.

Anything not mentioned here means we do not plan to include it in the next release, so please comment if I am missing something.

Once these items are completed, I plan to immediately create a draft release for beta test. We've changed quite a bit this round, and I expect there to be issues we have not discovered yet.

pliablepixels, 2015-10-22:
API: Confirmed we are at a good stopping point for 1.29
Cool ideas: no more PRs till 1.29 is released.

kylejohnson, 2015-10-23:
I'd like to get #1120 into this release.

SteveGilvarry, 2015-10-29:
@knnniggett all the details are in the pull I did, if you want me to document it and add to the release wiki entry can do.
https://github.com/ZoneMinder/ZoneMinder/pull/966

knnniggett, 2015-10-29:
@SteveGilvarry no, that's good enough. I just couldn't remember where you mentioned it.

pliablepixels, 2015-10-29:
That plugin rocks. I discovered it yesterday and migrated to it as well.

pliablepixels, 2015-11-02:
I'd also request we add this https://github.com/ZoneMinder/ZoneMinder/pull/1136 to 1.29 - it will help me implement previous/next and gapless event navigation

kylejohnson, 2015-11-02:
We gotta do a code freeze at some point!

onlyjob, 2015-11-03:
Maybe we could include ffmpeg-2.9 compatibility update #1137?

knnniggett, 2015-11-08:
I'm pasting this here so I can see what the markup looks like.

I used the following command to generate this list:
`github_changelog_generator --since-tag v1.28.1 --token [my token] ZoneMinder/ZoneMinder`

If you think one of the many options this tool supports would be helpful then please let me know.

# Change Log

## [Unreleased](https://github.com/ZoneMinder/ZoneMinder/tree/HEAD)

[Full Changelog](https://github.com/ZoneMinder/ZoneMinder/compare/v1.28.1...HEAD)

**Implemented enhancements:**

- Modify Frames table to not have a composity key \[$150 awarded\] [\#663](https://github.com/ZoneMinder/ZoneMinder/issues/663)
- Proposed: Enable purgewhenfull filter by default [\#332](https://github.com/ZoneMinder/ZoneMinder/issues/332)
- Change help text of source "Maximum FPS" and "Alarm Maximum FPS" to note that this should only be used on analog cameras [\#273](https://github.com/ZoneMinder/ZoneMinder/issues/273)
- Make zmpkg.pl 'better' [\#272](https://github.com/ZoneMinder/ZoneMinder/issues/272)
- Implement unit testing [\#266](https://github.com/ZoneMinder/ZoneMinder/issues/266)
- Migrate completely to C++ [\#265](https://github.com/ZoneMinder/ZoneMinder/issues/265)
- Enhance security by creating SELinux policies. [\#248](https://github.com/ZoneMinder/ZoneMinder/issues/248)
- ZoneMinder domain / website / email [\#225](https://github.com/ZoneMinder/ZoneMinder/issues/225)
- All of the Config table shouldn't be written in Configure::write [\#192](https://github.com/ZoneMinder/ZoneMinder/issues/192)
- Configuration options should be handled in the model, not the controller [\#191](https://github.com/ZoneMinder/ZoneMinder/issues/191)
- Travis CI automatic building [\#168](https://github.com/ZoneMinder/ZoneMinder/issues/168)
- Switch h264 feature branch to track the modern branch [\#167](https://github.com/ZoneMinder/ZoneMinder/issues/167)
- Add redhat specific files to source tree [\#115](https://github.com/ZoneMinder/ZoneMinder/issues/115)
- Update zmupdate to look at the github repo for the latest version [\#109](https://github.com/ZoneMinder/ZoneMinder/issues/109)
- Wiki Migration [\#108](https://github.com/ZoneMinder/ZoneMinder/issues/108)
- Update wiki page \(please?\) [\#91](https://github.com/ZoneMinder/ZoneMinder/issues/91)
- Exportable Camera Configs [\#86](https://github.com/ZoneMinder/ZoneMinder/issues/86)
- Translation stuff done, please pull in and test... [\#72](https://github.com/ZoneMinder/ZoneMinder/issues/72)
- Updated Hungarian translation [\#35](https://github.com/ZoneMinder/ZoneMinder/issues/35)
- Better ffmpeg Version Checking [\#31](https://github.com/ZoneMinder/ZoneMinder/issues/31)
- Scripts should only be loaded on the pages they're needed [\#15](https://github.com/ZoneMinder/ZoneMinder/issues/15)
- Replace PHP short tags [\#11](https://github.com/ZoneMinder/ZoneMinder/issues/11)
- Performing count\(\) in SQL is slower than in PHP [\#2](https://github.com/ZoneMinder/ZoneMinder/issues/2)
- Add the ability to specify the zm configdir at build time. [\#567](https://github.com/ZoneMinder/ZoneMinder/pull/567) ([knnniggett](https://github.com/knnniggett))
- add feature to extend preclusive zone alarm state by x frames [\#338](https://github.com/ZoneMinder/ZoneMinder/pull/338) ([m-bene](https://github.com/m-bene))

**Fixed bugs:**

- zmupdate.pl sort on sql files incorrect [\#1051](https://github.com/ZoneMinder/ZoneMinder/issues/1051)
- Streaming an Event as MPEG fails. [\#686](https://github.com/ZoneMinder/ZoneMinder/issues/686)
- Zoneminder Web UI No Longer Refreshes After Change [\#534](https://github.com/ZoneMinder/ZoneMinder/issues/534)
- Monster log messages [\#491](https://github.com/ZoneMinder/ZoneMinder/issues/491)
- zms crashes due to monitor not running. [\#468](https://github.com/ZoneMinder/ZoneMinder/issues/468)
- Flat Skins Export Functions missing $ [\#455](https://github.com/ZoneMinder/ZoneMinder/issues/455)
- unexpected } in console.php [\#354](https://github.com/ZoneMinder/ZoneMinder/issues/354)
- zmu fails silently when it cannot read zm.conf [\#339](https://github.com/ZoneMinder/ZoneMinder/issues/339)
- Flat skin needs window widths fixed [\#331](https://github.com/ZoneMinder/ZoneMinder/issues/331)
- ERR: Can't insert frame: Out of range value for column 'Score' at row 1 [\#319](https://github.com/ZoneMinder/ZoneMinder/issues/319)
- incomplete path to arp in zoneminder/skins/classic/views/monitorprobe.php causes probe code 127 failures [\#271](https://github.com/ZoneMinder/ZoneMinder/issues/271)
- Configuration gets broken in database after crash [\#239](https://github.com/ZoneMinder/ZoneMinder/issues/239)
- zmc crashes after a few minutes of usage in 1.26.4 [\#237](https://github.com/ZoneMinder/ZoneMinder/issues/237)
- Post shmget problem due to \*code\* not shmmax/shmall [\#226](https://github.com/ZoneMinder/ZoneMinder/issues/226)
- SQL error when "pre event image count" \>150 or so due to insufficient ZM\_SQL\_LGE\_BUFSIZ [\#222](https://github.com/ZoneMinder/ZoneMinder/issues/222)
- RTSP decoding errors in 1.26.4 \[$450 awarded\] [\#221](https://github.com/ZoneMinder/ZoneMinder/issues/221)
- 'asm' operand has impossible constraints [\#219](https://github.com/ZoneMinder/ZoneMinder/issues/219)
- ipv6 support [\#210](https://github.com/ZoneMinder/ZoneMinder/issues/210)
- Vivotek cameras, ultra slow FPS. [\#204](https://github.com/ZoneMinder/ZoneMinder/issues/204)
[... cut ...]
- Change default dB Engine from MyISAM to InnoDB [\#77](https://github.com/ZoneMinder/ZoneMinder/pull/77) ([knnniggett](https://github.com/knnniggett))
- Added build notes for CentOS & RHEL [\#76](https://github.com/ZoneMinder/ZoneMinder/pull/76) ([knnniggett](https://github.com/knnniggett))
- Update hu\_hu.php [\#37](https://github.com/ZoneMinder/ZoneMinder/pull/37) ([nagyrobi](https://github.com/nagyrobi))

\* *This Change Log was automatically generated by [github_changelog_generator](https://github.com/skywinder/Github-Changelog-Generator)*

pliablepixels, 2015-11-08:
One thing to note is that we need to correctly categorize bugs vs.
enhancement labels before we generate this. the current list has them
mixed due to how they were labelled.

knnniggett, 2015-11-10:
Just thought of something else. 

As part of the release notes for 1.29, I would like to make an end of life announcement for the following:
- Autotools (replaced by cmake)
- mobile skin (use zmNinja instead)
- XML skin (replaced by API)

I'd like ZoneMinder 1.29 to be the last version to support these items.

This is an invitation for discussion. Please let me know what you think.

kylejohnson, 2015-11-10:
@knnniggett - If that's the case, I say we remove the respective files from the repo, e.g. `web/skins/(mobile|xml)`

knnniggett, 2015-11-10:
Yes, exactly. Following the release of 1.29 and the end-of-life announcement, I'd like to remove all the files that make up the mentioned items. I do want to make sure we announce it sufficiently so we minimize the surprise factor. 

As far as I know, those of us doing active development have switched to cmake.

SteveGilvarry, 2015-11-10:
Any archs that don't support cmake?

pliablepixels, 2015-11-10:
if I remember correctly, removing them will break zmView Pro though (not sure but I think it does rely on some of these skins)

knnniggett, 2015-11-11:
@SteveGilvarry Archs or Distros? cmake is available in every Linux distro that I am aware of and is also available for Mac OS X, freebsd, and Solaris. Are there any other OS'es come to mind?  

@pliablepixels I think @kylejohnson would agree that the intent of the API is to replace current third party integration methods. We would expect the authors of any third party apps out there still using the XML skin to transition to using the API.

So the idea would be to make the end of life announcement, wait a reasonable amount of time for additional feedback, and then actually delete the items in question when we feel we have waited sufficiently.

SteveGilvarry, 2015-11-11:
@knnniggett that probably covers it, I was trying to think what we had seen people trying to run it on, be it OS and also architectures, but I suppose it is an OS thing and that is all the ones I have seen.

Also note I am having a look at the changelog generator to see if I can get it to only bring in pull requests on master.

knnniggett, 2015-11-12:
@connortechnology Can you speak to the current status of Freebsd development, particularly whether or not the build process has been migrated to cmake?

connortechnology, 2015-11-12:
I used cmake the last time I tried to compile on FreeBSD. It works fine.

connortechnology, 2015-11-12:
The FreeaBSD ports thing uses autoconf, but I'm sure they don't really care.  Their build process has a lot of alternative infrastructure.

Honestly, I only care about it basically compiling on freebsd, not how easy it will be.  Anyone building this on freebsd will have to know what they are doing.

pliablepixels, 2015-11-15:
@knnniggett - folks on the forum seem to be commenting that neither scilinux nor centOS7 supports the image scale API that  #1000 requires - I'm not sure that's correct, but I'm hearing more of it, if you'd like to take a look at least on CentOS before merging. Thread in question: https://forums.zoneminder.com/viewtopic.php?f=32&t=23645&p=90523#p90523 --> both BiloxiGeek (SciLinux) and Abi (CentOS7) seem to be saying this

knnniggett, 2015-11-15:
All they have to do is make sure the php-gd package is installed.

Thanks,

Andy

No Trees were killed in the sending of this message.  
However, a large number of electrons were terribly inconvenienced.

Date: Sun, 15 Nov 2015 09:25:24 -0800

pliablepixels, 2015-11-15:
Doesn't seem to be the case @knnniggett  - looks like imagescale() is only available in PHP 5.5 while CentOS 7 seems to have 5.4

knnniggett, 2015-11-15:
Then that is a problem. We will need to do something other than call imagescale when php < 5.5

Thanks,

Andy

No Trees were killed in the sending of this message.  
However, a large number of electrons were terribly inconvenienced.

Date: Sun, 15 Nov 2015 11:53:53 -0800

knnniggett, 2015-11-15:
A quick google suggested that the imagecopyresized function (part of php-gd) can be used instead of imagescale

http://php.net/manual/en/function.imagecopyresized.php

Suggested here at the bottom:
http://stackoverflow.com/questions/27813479/imagescale-function-in-php

Thanks,

Andy

No Trees were killed in the sending of this message.  
However, a large number of electrons were terribly inconvenienced.

Date: Sun, 15 Nov 2015 11:53:53 -0800

SteveGilvarry, 2015-11-18:
@knnniggett I would recommend --no-issues as a flag for change log, this removes the spam issues and results in the Pull requests only which are the actual changes. I just did a PR on the generator to include a filter for release branch as current system brings in all PR's including ones on video and other branches. https://github.com/skywinder/github-changelog-generator/pull/305
Then I would run it like this to create the 1.29.0 changelog file.
```github_changelog_generator -t <Token>  --no-issues --future-release 1.29.0 --release-branch master ZoneMinder/ZoneMinder```

connortechnology, 2015-11-18:
I would like to propose that we name or in some way commemorate @manupap1 with this release.  He made some very significant contributions during his time here.

knnniggett, 2015-11-18:
Excellent idea. I second the motion.

knnniggett, 2015-11-21:
@SteveGilvarry --release-branch does not appear to be a valid parameter for version 1.9.0

SteveGilvarry, 2015-11-21:
Yes I have a PR on the project to add it just have discussion about
style/cleanup. How about I run it once we are ready? I can PR change log
now so you can see it?

knnniggett, 2015-11-21:
Yeah, would you mind generator one now just so I can see it?

With help, I finally fixed #1153 . I feel as if a great weight has been lifted.

knnniggett, 2015-11-22:
DRAFT

#Fade to Black (Proposed release title)

###In Memory of  Emmanuel Papin

The ZoneMinder team would like to dedicate this release to Emmanuel Papin.  Emmanuel was a significant contributor to this project and even went on to write a license plate plugin. You can view his work here:  https://github.com/manupap1/libzoneminder-plugin-openalpr

It is with great sorrow that I must report his passing on August 19, 2015. He will be missed.

###EOL Announcement

The following features are planned to be removed from the ZoneMinder project in a future release:
- Autotools – Build using cmake instead
- Mobile Skin – Use a mobile app, such as [zmNinja](https://github.com/pliablepixels/zmNinja)
- XML Skin – Use the new API instead

###Significant New Features and Changes

The ZoneMinder API! This is now the supported way for third party tools and apps to communicate with ZoneMinder. Please contact us if you find anything lacking in the API.

Referred to by us as MultiServer, ZoneMinder now supports the display of cameras attached to other ZoneMinder servers from a single web console. Previously, something similar could be done by adding a camera with a url that pointed to another ZoneMinder server. While this worked, it was inefficient, as processing and storage were duplicated across both servers. The new MultiServer feature accomplishes the same thing, but leaves the processing and storage to just one server. This allows ZoneMinder to scale better.

The option USE_DEEP_STORAGE has been hidden. This option has been enabled as the default for many years. Recently, it was identified that turning this option off resulted in the near immediate deletion of events from the disk. Not good. Consequently, this option can no longer be modified from Options. In the unlikely event that you have a working system not already using deep storage, please contact us so we can get that enabled without loss of your data.

ZoneMinder no longer uses php short open tags.

ZoneMinder has limited ONVIF discovery support, but it must be enabled at build time. We could use someone to build and test this on different makes and models of cameras.

###Changelog

[insert output from changelog generator here]

onlyjob, 2015-11-22:
Sad news about Emmanuel. :( Is there a public obituary?

knnniggett, 2015-11-22:
@onlyjob We received the following post in our forum, notifying us of Emmanuel, just a few days ago:
https://github.com/ZoneMinder/ZoneMinder/pull/1160#issuecomment-157692925

I have tried, but have not been able to find any additional information through online searches.

aplocher, 2015-11-30:
Is it considered a faux pas to mention the "EOL Announcement" immediately after mentioning Emmanuel? (or is it a faux pas to point that out... d'oh). Sounds like a great dev, sad news...

knnniggett, 2015-11-30:
@aplocher Yes, I had considered that, but was waiting to see if anyone besides myself noticed. I don't want to move it to the end because it won't be read. Perhaps I could call it something other than "EOL".

SteveGilvarry, 2015-11-30:
Feature Deprecation, which you would do this release to advise people to
stop using.

knnniggett, 2015-11-30:
Ah, thanks @SteveGilvarry . The wording has been updated.

knnniggett, 2015-12-04:
The past holiday weekend kind of put things on the breaks.
I'm trying to reignite motivation on a 1.29.0 release, myself included.

As I understand it, we've got these three issues to work on:
- Test Multi-Server and document
- Fix #1167 
- We need to investigate libvlc crashes, as it has become widespread, rather than isolated.

SteveGilvarry, 2015-12-04:
#1179 put in for that libvlc issue. 

#1176 fixes the version issues in #1167

SteveGilvarry, 2015-12-19:
All I want for Christmas is my 1.29 proposed release.

onlyjob, 2015-12-19:
Yeah, and my children will have no Christmas presents this year if you don't release. :wink:

knnniggett, 2015-12-19:
With #1190 merged, we are down to just one last major feature to address before release - multi-server.

pliablepixels, 2015-12-19:
I've buttoned up the API security issue for now  - so if you want to you can merge it. That way folks who really don't want the API have an option https://github.com/ZoneMinder/ZoneMinder/issues/1195 -- your call to include or not

knnniggett, 2016-01-01:
Alright, we finally made it. I am going to bump master to 1.29.0,  spawn a new 1.29.0 branch, and then create a draft release.  

On a side note, perhaps someone can explain to me how new documentation, placed under our /docs folder, makes it over to http://zoneminder.readthedocs.org . Is there a manual process what must be done?

kylejohnson, 2016-01-01:
Automated via webhook

pliablepixels, 2016-01-01:
Awesome! look forward to it!

knnniggett, 2016-01-01:
A draft release has been created and is available on the Releases Page. @SteveGilvarry when you've got a moment, can you add your github_changelog_generator goodness?

knnniggett, 2016-01-01:
Well this sucks. Because our project contains submodules, I have uploaded a tar.gz containing ZoneMinder + submodules, but I can't seem to make the other links, which Github auto-created, go away.

I'm searching for an answer or workaround, but so far I've not found anything.

pliablepixels, 2016-01-01:
I don't think there is a workaround. We should probably move to git subtree instead - that should take care of this problem.

kylejohnson, 2016-01-01:
Is it worth announcing and releasing the updated website, along with a blog post to match?

http://zoneminder.github.io is the current dev snapshot of the website.

knnniggett, 2016-01-01:
@kylejohnson Sounds fine to me. Feel fee to edit the release notes and add your comments.

pliablepixels, 2016-01-01:
@kylejohnson - absolutely. Better to start soon and iterate quick. Keeps the tempo going! The image should be removed (not licensed) and the Jumbotron reverted to black on grey. I'll do a PR later tonight if you don't already do it.

pliablepixels, 2016-01-02:
@kylejohnson - https://github.com/ZoneMinder/zoneminder.github.io/pull/5

SteveGilvarry, 2016-01-02:
@knnniggett I saved it as a draft as the other option looked like it would do a new release, you should see the PR's listed, and can publish. I changed headings to ## so I could use ### for a subheading that was in the log.
Also my PR for the changelog can be merged in to include it all in CHANGELOG.md

knnniggett, 2016-01-02:
Merged #1172 . Sorry, I didn't notice that pr was directed at me.

I just discovered that , when you save a release as a Draft, rather than Publish it, The Public cannot see either the modified version or even the original unedited (pre)release. It seems to just completely disappear from the Releases page. 

I am going to create Release Candidate 2 in a day or so as there are already some changes we need to add to that 1.29 branch.

In order to update the tag and commit hash, it looks like I've got to delete and then recreate the entire pre-release. Following that, feel free to modify the Changelog only if you want to. I don't think it needs to be 100% perfect.

SteveGilvarry, 2016-01-02:
The other option made it sound like I was doing another release, but I guess it just republishes that existing one. I will update my PR on github-changelog-generator so you can also grab that and run it, they wanted me to fix some formatting but I haven't had time.
But also happy to run everyday and put in a PR on CHANGELOG.md, then you just grab what you want when ready to release. I suspect it might keep 1.29.0rc1, then 1.29.0rc2 etc, then 1.29.0, won't know until we release them. Bit of a chicken or egg problem with Changelogs as updating them is also a PR.

```
bundle exec github_changelog_generator -t [InsertToken] -o ../CHANGELOG.md --no-issues  --release-branch master ZoneMinder/ZoneMinder

knnniggett, 2016-01-04:
I've merged #1222 into the 1.29.0 branch, and will test one more time, hopefully this evening.
Assuming I don't run into any new issues, I will create Release Candidate 2.

connortechnology, 2016-01-04:
Do you want me to do a PR for not showing the server column on the console for the non-multi-server case?  I kinda think I should.

knnniggett, 2016-01-04:
Yeah, that's probably a good idea.

[26 more replies]

closed 2016-02-04 by knnniggett