ZoneMinder/zoneminder #1131: Upgraders will need db permission for INDEX for zmupdate.pl to apply 1.28.108.sql

SteveGilvarry, 2015-10-31:
@knnniggett @connortechnology 
Need to add to the readme for the distros that upgraders need INDEX for zmuser to run the 108 sql script which creates an Index.
```grant select,insert,update,delete,lock tables,alter,create,index on zm.* to 'zmuser'@localhost identified by 'zmpass';```

knnniggett, 2015-10-31:
Really? That's annoying, considering one can run zm_create.sql w/o that permission.

It seems one can create a new table with indexes, but one cannot create just the index w/o this permission enabled according to mysql:
`If you have the CREATE privilege for a table, you can include index definitions in the CREATE TABLE statement. `

knnniggett, 2015-11-04:
@onlyjob 
@barjac 

Just making sure you guys see this

barjac, 2015-11-04:
I had a further comment but it involved a patch attachment which this crazy board offers to attach and then refuses saying it has to be an image or text - so I change it to .txt and it is again rejected saying it has to be png gif or jpg and that I need write permission to the repository to attach a document of any sort. Paranoid or what!

SteveGilvarry, 2015-11-19:
I have some ideas, and thought I would run them past people before I even try it. Or if someone good with perl wants to take it on just put hand up and assign issue to yourself. @connortechnology mentioned something about not using zmdbuser the other day and I thought this would be a good compromise.
1. Modify zmupdate.pl to check db user permissions to ensure they have the required set including create and index.
2. When they don't notify them and give them to option to stop and fix, or ask user to enter the details of a user that does have correct permissions, i.e. root. And check the user does before continuing.

I would also like to propose we stop supporting DB upgrades from ancient versions. zmupdate is almost half updates for old versions.
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmupdate.pl.in#L494-L1004
Plus 40 odd sql files.

barjac, 2015-11-19:
OK this brings me back to what I was going to say when I got so frustrated with the board.
For as long as I can remember we have been applying this patch (or similar) to zm_create.sql and it has never caused an issue. (using a paste bin this time).
https://paste.kde.org/pgpmcj9hn
Will be updated of course now with INDEX and CREATE.

SteveGilvarry, 2015-11-19:
I suspect if we put that in then other distros would patch it out for setting a default username and password on a db account. 

```diff
diff -ur ZoneMinder-4332_orig/db/zm_create.sql.in ZoneMinder-4332/db/zm_create.sql.in
--- ZoneMinder-4332_orig/db/zm_create.sql.in	2014-09-03 20:59:42.000000000 +0100
+++ ZoneMinder-4332/db/zm_create.sql.in	2014-09-05 13:57:09.762570208 +0100
@@ -643,6 +643,10 @@
 --
 -- Apply the initial configuration
 --
+USE mysql;
+GRANT SELECT,INSERT,UPDATE,DELETE,ALTER ON zm.* TO 'zmuser'@localhost IDENTIFIED BY 'zmpass';
+USE `@ZM_DB_NAME@`;
+--
 -- This section is autogenerated by zmconfgen.pl
 -- Do not edit this file as any changes will be overwritten
 --
```

barjac, 2015-11-19:
Yes, probably, but in our case where my objective was to make ZM as easy as possible for users to install and use, it has worked well without any complaints.
Just thought I would add it to the discussion ;)

knnniggett, 2015-11-19:
@barjac I think we are going to have to agree to disagree. That patch isn't going to meet everyone's requirements, and I don't think any of us are willing to make a change that breaks other packages. I think a better way to handle the mysql account, that works for everyone, is to make this part of an Installation Wizward which steps the user through this. See #12. In my head, I have envisioned this as something similar to Drupal's setup.php if you are familiar with that.  Since we always seem to have bigger issues though, this has not been a priority.

knnniggett, 2015-11-19:
I tend to agree with all of recent comments from @SteveGilvarry 

Another, related issue involves the ever increasing credentials requirements of the zmuser account. To really be secure, the zmuser account should only have the minimum permission necessary for normal day-to-day operations.

I would propose adding a second maintenance account, which has been granted ALL (?) permission. This maintenance account would not be accessible from the web portal, and only accessible from zmupdate.pl.  

I guess you could this of this account as a replacement to the debian maintenance account (Redhat distros have never had an account like this).

Are there any objections?

kylejohnson, 2015-11-19:
I am mixed here.  Of course least-privilege is good, but there has to be a trade off with complexity.  

If the goal is to limit potential damage, then I would argue that if you already have `UPDATE, DELETE` privileges, you can do just as much damage as you could with the additional `INDEX, ALTER` privileges.  If there is already a malicious user on the system, you are already in a heap of trouble.

For the sake of keeping things simple, I think my vote is for `ALL PRIVILEGES ON zm.* TO zmuser@localhost`, as opposed to constantly handling edge cases.

barjac, 2015-11-19:
I was certainly not proposing that patch as a patch for the project - just showing how we have done it.
Thinking about it, that could be incorporated into our zmsetup script to remove the patch altogether.
http://svnweb.mageia.org/packages/cauldron/zoneminder/current/SOURCES/zmsetup?view=markup

connortechnology, 2015-11-19:
I don't think there is a definite answer, but we can improve in baby steps.  

People building from source assume a certain level of capability, so for them I think it's about detection and giving a nice error message.  So we need to detect whether we have the permission or not.

For debian based packages, I think we should just run the upgrade as the debian db maintainer user, bypassing it all.  zmupdate needs to be run as root anyways.

I don't know about other distros.

connortechnology, 2015-11-19:
I don't really see any reason to remove the updates for old versions.

knnniggett, 2015-11-19:
For comparison, I checked the mysql permissions on some of my production boxes and here is what I found:

Drupal -> GRANT SELECT, INSERT, UPDATE, DELETE, CREATE, DROP, INDEX, ALTER

Gallery2 -> GRANT SELECT, INSERT, UPDATE, DELETE, CREATE, DROP, REFERENCES, INDEX, ALTER, CREATE TEMPORARY TABLES, LOCK TABLES, EXECUTE, CREATE VIEW, SHOW VIEW, CREATE ROUTINE, ALTER ROUTINE

Snort -> GRANT SELECT, INSERT, UPDATE, DELETE

Zabbix -> GRANT ALL PRIVILEGES

Mythtv -> GRANT ALL PRIVILEGES

Looking at the data, I tend to agree with @kylejohnson that maybe the best course of action, moving forward, is to just update our documentation to grant all permission to zmuser.

I also think it a good idea to check the permissions when running zmupdate.pl and output a human-friendly response that the end user needs to update their account permissions before continuing.

I can add this to my task list, following the 1.29 release, if we agree these two steps are a good next step.

SteveGilvarry, 2015-11-19:
@knnniggett I think the check needs to be in 1.29 as it is adding these permissions and when we are likely to get hammered from people not doing it.

@connortechnology only reason is housekeeping and establishing end of life for older versions.

closed 2017-03-18 by SteveGilvarry