ZoneMinder/zoneminder #1134: Failed to start zoneminder.service: Unit mysqld.service failed to load: No such file or directory.

dreaddy, 2015-11-01:
Ubuntu 15.10

Long story short. 

I installed from source. That did not go well. Purged/removed and installed from ubuntu repo.
Been plagued by the following error ever since.

Failed to start zoneminder.service: Unit mysqld.service failed to load: No such file or directory.
OR
Failed to start zoneminder.service: Unit httpd.service failed to load: No such file or directory.

Does anyone know to fix this?

knnniggett, 2015-11-02:
It's telling you it can't start zoneminder because it depends on apache/mysql and it could not start those services. You should be following the Ubuntu 15.10 "easy way" documentation found in the Wiki.
Since this does not describe a bug, please post questions to the user forum for further help.

closed 2015-11-02 by knnniggett

dreaddy, 2015-11-02:
"You should be following the Ubuntu 15.10 "easy way" documentation found in the Wiki"

Installed the latest development branch. Ubuntu 15.10 "easy way" documentation is not very useful.

knnniggett, 2015-11-02:
See our posting guidelines:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Bugs and feature requests belong in our github forum. All other discussions are to be made in our IRC channel or user forum.