ZoneMinder/zoneminder #1140: API implementation of getDiskPercent() is killing large filesystems running ZM

pliablepixels, 2015-11-03:
I've been receiving reports that zmNinja often does not show "disk space" correctly. Folks running it on large systems (10+ cameras and many events) also emailed me saying when they try to see Disk Space in zmNinja it seems to almost kill ZM.

On further investigation, it looks like getDiskSpace implementation here: https://github.com/ZoneMinder/ZoneMinder/blob/master/web/api/app/Controller/HostController.php#L75 is calling a ``du -s0`` for each monitor, while the PHP code in ZM calls PHP's get_disk_space which uses a stat approach.

User jzaw at IRC forums ran the test script  on a 10 monitor system and reported his CPU load went very high and it took 190 seconds to complete. 

Test script:  https://gist.github.com/pliablepixels/0f2223a1ec93cd4d5b08
Test result: https://gist.github.com/pliablepixels/3433d7bc5568a5a96958

In summary, we need to rework this API implementation. I've disabled it in zmNinja for now.

kylejohnson, 2015-11-03:
Yup.  When I put that code in, I knew it would have to be improved.

connortechnology, 2015-11-03:
I kinda think we should have a daemon that automatically updates a db field or something every few minutes.

Honestly, I think calculating the size of any of my events dirs would bring my machines to their knees. 

df is about the only reasonable measurement to take.

  Alternatively, maybe we need to calculate the size of each and every event, and keep totals in the db.

Doing something like du on our event dir will kill any machine.  

Isaac

pliablepixels, 2015-11-03:
I dug in a little bit earlier this morning and here is what I think is the case:

a) The API uses "du" which does a file traversal.
b) The PHP code in ZM does an equivalent of "df" which actually uses statvfs which is retrieving information from the kernel's view of the filesystem, I think at the inode level which s very fast. https://github.com/ZoneMinder/ZoneMinder/blob/b2b52a364a69ecd1002edea6df410739d77e4940/web/includes/functions.php#L1640

So b is much faster (maybe a little inaccurate as its not really traversing files in real time) and that might be the right approach to plough into the APIs.

alager, 2015-11-04:
My two cents, would be to have the API use method b.  This does several things:
1) consistent numbers between API and ZM console.
2) could reduce code through refactoring
3) no need to add a daemon

Velociraptor85, 2015-12-03:
API (https://github.com/ZoneMinder/ZoneMinder/blob/master/web/api/app/Controller/HostController.php#L75) call relies on 'du' calls maybe as a quick fix one could try the 'ls' solution from https://stackoverflow.com/a/29826677

there are also plain php functions which might help https://stackoverflow.com/a/21409562

the difference to the php in the ZM (https://github.com/ZoneMinder/ZoneMinder/blob/b2b52a364a69ecd1002edea6df410739d77e4940/web/includes/functions.php#L1640) seems to me that you want to have an array of filesizes for each Monitor and I am not sure if that would work with the PHP funktions "disk_total_space" and "disk_free_space" because according to https://secure.php.net/manual/en/function.disk-free-space.php they only "show available space on filesystem or disk partition"

2cent

closed 2017-03-18 by SteveGilvarry