ZoneMinder/zoneminder #1141: Migrate forums to new host

kylejohnson, 2015-11-03:
Currently migrating the forums from their current home, to a new one.
In addition, moving them from zoneminder.com/forums to forums.zoneminder.com

kylejohnson, 2015-11-03:
https://forums.zoneminder.com/ is up.
Please log in, try to post, etc.  If things look good, I'll redirect zoneminder.com/forums to https://forums.zoneminder.com/ and then call it good.

kylejohnson, 2015-11-03:
Got an 'A' rating with SSL Labs.  Not the A+ I'm hoping for, but it'll do for now:
https://www.ssllabs.com/ssltest/analyze.html?d=forums.zoneminder.com&s=52.6.136.15&latest

[image]

kylejohnson, 2015-11-03:
I've increased both the maximum attachment size, and maximum attachment quota, essentially re-enabling users to upload attachments.

knnniggett, 2015-11-03:
How configurable is the search engine? Do you have control over what words it thinks are "too common"? It won't let you search on "mysql" for example, which totally blows goats.

I was able to successfully create and then delete a post on forums.zoneminder.com.
However, I am probably not a good example as my account has elevated privileges.

@pliablepixels you mind creating a post in the new forum?

pliablepixels, 2015-11-03:
@knnniggett https://forums.zoneminder.com/viewtopic.php?f=13&t=23635

kylejohnson, 2015-11-03:
I'm currently doing an upgrade of phpbb3, so there may be issues for the new few minutes.

kylejohnson, 2015-11-03:
Upgrade is finished.

kylejohnson, 2015-11-03:
Any other maintenance / things to do before I setup the redirect?

kylejohnson, 2015-11-03:
> How configurable is the search engine? Do you have control over what words it thinks are "too common"? It won't let you search on "mysql" for example, which totally blows goats.

Slightly configurable.  For example, one can select a different search backend.  Currently it is using phpbb native, though we could change it to mysql native, or to sphinx (which would probably be the best).

We should explore this in another issue.

knnniggett, 2015-11-03:
The new phpbb3 is working for me. 
Thanks for fixing the attachments

Some of us such as myself and @pliablepixels and bbunge could really use the ability to post stickies. Don't know if you can enable that easily, or if we should fix that some other day.

kylejohnson, 2015-11-03:
Could we please confirm that old URls now redirect to the new site?

e.g. http://www.zoneminder.com/forums/viewtopic.php?f=9&amp=&t=23516 -> https://forums.zoneminder.com/forums/viewtopic.php?f=9&amp=&t=23516

kylejohnson, 2015-11-03:
Doing a bit more maintenance, may cause some downtime.

kylejohnson, 2015-11-03:
It seems that I've broken the forums for the next 24 hours due to a botched DNS updated.  Some users will still have access, while others won't.

alager, 2015-11-04:
The search seem mediocre. :(
It doesn't appear to like numbers at all.  eg. tried all variants of 1.28.107 (double quotes, single quotes, back slashes) and got 0 results, yet there is a thread: "ZM 1.28.107 problems after upgrade "

kylejohnson, 2015-11-04:
Try the search for 1.28.107 now.  I've changed the search backend.

alager, 2015-11-05:
much better!

kylejohnson, 2015-11-05:
Going to close this out.  I'd say the migration is a success.
Performance is better, we're on a new release, search is better, and we've got SSL.

Please open up another ticket for any more forum-related issues (such as a better theme / skin @pliablepixels )

closed 2015-11-05 by kylejohnson