ZoneMinder/zoneminder #1146: Unable to generate video, check /usr/share/zoneminder/www/events/.../ffmpeg.log

dreaddy, 2015-11-05:
Re-opened previously closed issue #1126 

Fix for the following log ERR experienced using ubuntu 15.10

    zmvideo[10668]: INF [Creating video file Event-1028-r1-s1.avi for event 1028]
    zmvideo[10668]: ERR [Unable to generate video, check /usr/share/zoneminder/www/events/1/15/10/04/22/38/37/ffmpeg.log for details]

connortechnology, 2015-12-31:
So... I don't think the \ in front of $command is the right thing to do.  You seem to be complaining that there is an error in your logs... but the error is telling where to look to find out what the error is.  You are going to have to go look in /usr/share/zoneminder/www/events/1/15/10/04/22/38/37/ffmpeg.log to find out.

dreaddy, 2015-12-31:
The ffmpeg.log wasn't generated either. Adding the escape(\) character was a quick solution.

closed 2017-03-18 by SteveGilvarry