ZoneMinder/zoneminder #1151: RTSP monitors don't not work with IPv6

schrorg, 2015-11-07:
I am trying to build a IPv6-only system with about 100 cameras. That means I would like to have ZoneMinder connect to the cameras over IPv6. With ffmpeg and libvlc it would be no problem. I.e. #252 already solves this for ffmpeg.

But I would like to use the native RTSP code from ZoneMinder. And that seems to only support IPv4. The problem is, that the functions in `src/zm_comms.{h,cpp}` and some other files use `gethostbyname(3)`, which is not IPv6 compatible:

> 11/07/15 14:39:01.030891 zmc_m10[1652].DB1-zm_logger.cpp/234 [LogOpts: level=DB5/DB5, screen=OFF, database=INF, logfile=DB5->/var/log/zm/zm_debug.log.01652, syslog=WAR]
> 11/07/15 14:39:01.030951 zmc_m10[1652].DB1-zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
> 11/07/15 14:39:01.031433 zmc_m10[1652].DB1-zm_monitor.cpp/2726 [Got 1 monitors]
> 11/07/15 14:39:01.031449 zmc_m10[1652].DB1-zm_monitor.cpp/2759 [Got 1 for v4l_captures_per_frame]
> 11/07/15 14:39:01.031470 zmc_m10[1652].DB2-zm_camera.cpp/39 [New camera id: 10 width: 1280 height: 720 colours: 3 subpixelorder: 6 capture: 1]
> 11/07/15 14:39:01.033006 zmc_m10[1652].DB1-zm_thread.cpp/241 [Creating thread]
> 11/07/15 14:39:01.033031 zmc_m10[1652].DB2-zm_rtsp.cpp/198 [RTSP Local SSRC is 2112cd0c]
> 11/07/15 14:39:01.033040 zmc_m10[1652].DB1-zm_thread.cpp/281 [Starting thread]
> 11/07/15 14:39:01.033119 zmc_m10[1659].DB2-zm_thread.cpp/253 [Invoking thread]
> 11/07/15 14:39:01.033202 zmc_m10[1652].DB1-zm_thread.cpp/302 [Started thread 1659]
> 11/07/15 14:39:01.033232 zmc_m10[1652].DB2-zm_image.cpp/168 [Blend: Using SSE2 fast blend function]
> 11/07/15 14:39:01.033246 zmc_m10[1652].DB2-zm_image.cpp/204 [Delta: Using SSSE3 delta functions]
> 11/07/15 14:39:01.033252 zmc_m10[1652].DB2-zm_image.cpp/248 [Deinterlace: Using SSSE3 delta functions]
> 11/07/15 14:39:01.033255 zmc_m10[1652].DB2-zm_image.cpp/261 [Image buffer copy: Using SSE2 aligned memcpy]
> 11/07/15 14:39:01.033259 zmc_m10[1652].DB3-zm_image.cpp/268 [Setting up static colour tables]
> 11/07/15 14:39:01.033290 zmc_m10[1652].DB1-zm_monitor.cpp/388 [monitor purpose=1]
> 11/07/15 14:39:01.033295 zmc_m10[1652].DB1-zm_monitor.cpp/396 [mem.size=138241760]
> 11/07/15 14:39:01.033314 zmc_m10[1652].DB1-zm_monitor.cpp/541 [Unable to map file /dev/shm/zm.mmap.10 (138241760 bytes) to locked memory, trying unlocked]
> 11/07/15 14:39:01.033322 zmc_m10[1652].DB1-zm_monitor.cpp/544 [Mapped file /dev/shm/zm.mmap.10 (138241760 bytes) to locked memory, unlocked]
> 11/07/15 14:39:01.034041 zmc_m10[1659].ERR-zm_comms.cpp/201 [gethostbyname( host.name.of.my.camera ), h_errno = 4]
> 11/07/15 14:39:01.034726 zmc_m10[1659].FAT-zm_rtsp.cpp/238 [Unable to connect RTSP socket]

My test camera (substituted by "host.name.of.my.camera" in the above log output) has only a AAAA record, no A record.

With the great help of @ajobs we rewrote some of the code (actually, he did most of the work) to use IPv6 compatible mechanisms instead. We tested it for about a month now on two cameras with no problems.

I will do a pull request shortly.

knnniggett, 2015-11-07:
This sounds familiar, especially gethostbyname. I wonder if we've already got an issue open, related to this.

schrorg, 2015-11-07:
There was #210

SteveGilvarry, 2015-12-03:
Thoughts on using external libraries? http://www.boost.org/doc/libs/1_59_0/doc/html/boost_asio.html
I have no view on boost or any other library, but I think we spend time looking after code that is implemented better elsewhere.

connortechnology, 2015-12-03:
In general I like the idea of using external libraries.  In doing so, I think we need to avoid ones like ffmpeg and libav that are nightmares to track.  So ideally we want stable libraries that are generally available everywhere.
Boost is a good candidate.  I would like to see us use more c++-isms in our code, and re-inventing basic things like connecting to a server and requesting a stream is not a great idea.  So I'm all for using Boost.

knnniggett, 2015-12-03:
Since CentOS 6 can be the limiting factor when it comes to implementing something new, I checked the version of boost in the CentOS 6 repo. It is at version 1.41.0.  I hope this version is new enough if we decide to use it.

schrorg, 2016-05-15:
Fixed by #1152

closed 2016-05-15 by schrorg