ZoneMinder/zoneminder #1154: on slow ipcam "Open video took more than 10 seconds." error.

far5893, 2015-11-08:
I have a ipcam that take 12 seconds to start working video streaming from request (measured  with vlc with same rtsp:// string) and obviously log give me this error and zoneminder refuse to show me video.

There is a place where i can change this timeout in zoneminder ? I can try to recompile if necessary.

knnniggett, 2015-11-08:
Yes, that value is in the code somewhere, but I don't recall where it is exactly (and 10 is just too common of a value to search on).

We need to know which remote method you are using: liblv, ffmpeg, or Remote/RTSP.

Also, please provide all relevant error messages from the log. It should show which .cpp file the error is being generated from.

far5893, 2015-11-08:
zmc_m7	20476	ERR	Open video took more than 10 seconds.	zm_ffmpeg_camera.cpp 	413
zmc_m7	22382	ERR	Unable to open input rtsp://admin:admin@192.168.170.123:10554/tcp/av0_0 due to: Success	zm_ffmpeg_camera.cpp	264

far5893, 2015-11-08:
Ok there is a ffmpeg timeout in Options -> Image , now all its working , thank 
You can close.

closed 2015-11-09 by knnniggett