ZoneMinder/zoneminder #1155: APIs need to honor user roles

pliablepixels, 2015-11-10:
As of today, if you are using APIs, as long as you have a valid login, you have full access. We need to tie this into the user permissions so the APIs don't return any more data than the role allows.

mnoorenberghe, 2017-03-18:
What's the policy on closing issues? I notice many issues that had fixed merged weren't closed. Are they intentionally left open until release and then sometimes forgotten about? I think in many cases it's because the committer didn't include "fixes" before the issue number and then nobody came back to close it.

SteveGilvarry, 2017-03-18:
Just time and resources. I actually said just 2 days ago to the other team members that I was going to start closing out a few to get numbers down. So if you want to help flagging them I will follow behind to close, as well as reviewing them myself. Hopefully no one gets offended by me closing their issue, nothing personal.  Some of these should be in the forums as opposed to actual bugs or development features.

closed 2017-03-18 by SteveGilvarry