ZoneMinder/zoneminder #1158: ffmpeg libraries not found in Fedora 23 with zoneminder 1.28.1-4 (i686)

jplx, 2015-11-12:
I have Fedora 23 and compiled zoneminder-1.28.1-4.fc23.src.rpm
It is working OK with non-ffmpeg cameras like Foscam FI8905W or FI8918W.
I am trying to connect to FI9805W, which is ffmpeg based. I am getting the following error in the log:
First:   'zmc -m 3' starting at blahblah...
Then the error:   You must have ffmpeg libraries installed to use ffmpeg cameras to monitor 3
zm_monitor.cpp   Line 2684
I have the following ffmpeg packages installed:
ffmpeg and ffmpeg-devel 2.8.1-1.fc23
ffmpeg-libs 2.8.1-1.fc23
gstreamer-ffmpeg 0.10.13-15.fc22
The ffmpeg-libs components are installed at /usr/lib  (libavcodec, libavfilter, libavformat, libavresample, libavutil, libpostproc, libswresample, libswscale).
Is there a missing link between the location where Fedora 23 installs the ffmpeg library and where zoneminder expects it? Any other clue? Thank you for your help!

knnniggett, 2015-11-12:
Ah, so Fedora 23 is out. I need to update zmrepo.

> You must have ffmpeg libraries installed to use ffmpeg cameras to monitor 3

This error occurs when you have not built zoneminder with the ffmpeg headers installed prior to building. You can't install ffmpeg after you have already built zoneminder. You can also get this error if the _without_ffmpeg macro is defined in the rpm spec file.

To resolve this, you can either wait until I've got Fedora 23 support added to zmrepo, or you can try to rebuild a Fedora 22 src.rpm from zmrepo, following the rpm instructions shown in README.md.

I can't give you a more precise answer, because you have not told us exactly how you built the package you are using now.  

Since this issue does not describe a bug, I'm going to close this out. If you need additional help, you are welcome to continue this discussion in the user forum or our IRC channel.

closed 2015-11-12 by knnniggett

knnniggett, 2015-11-12:
Oh, and slightly related.... I tend to not notice when new versions of Fedora are released. Feel free to get my attention when a release date approaches.

jplx, 2015-11-12:
I will try to let you know about new Fedora releases.
The src.rpm file I used was found at: http://rpmfind.net/linux/rpm2html/search.php?query=zoneminder
Looking at the html file (left of web page), it does not require ffmpeg! The same thing for all the other compiled versions 1.28.1-4. Previous versions shown on the same page do require ffmpeg.
I will follow your advise and try to use the 1.28.1-1,fc22.src.rpm from your zmrepo.
I will try to change the version to something with fc23 by using rpmrebuild.
Question: after installing zoneminder and updating the zm.conf file (user, pwd), should I run zmupdate.pl --version=<from version>? I am asking because I am downgrading as opposed to updating.
Thank you for your help.

SteveGilvarry, 2015-11-12:
@jplx ffmpeg is not in fedora official repos so they cannot use it, end result zoneminder that doesn't work. ffmpeg is patent encumbered is the reason given for it not being available. Lots of multimedia applications are affected.

jplx, 2015-11-15:
I downloaded zoneminder-1.28.1-1.fc22.src.rpm from the zmrepo. In the spec file, I changed community-mysql-devel to mariadb-devel (because that it the one installed on the system I use for this). There were two "BUILD requires" that could not be found on the Fedora repositories but you had given them to me last year: perl-ASTRO-Suntime and perl-X10-ActiveHome. They installed fine in Fedora 23. After those changes, the zoneminder source file compiled successfully, including ffmpeg.
I can see the non-ffmpeg cameras but not the ones with ffmpeg. Here is the error message in the log:
FI9821P: 1000 - Capturing at 71.43 fps
Unable to open input rtsp://jplx:my_password@192.168.1.28:88/VideoMain due to: Operation now in progress...../home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp......264
I do not see ethernet traffic (green LED's blinking) on the hub, i.e. there is no transfer of data between the camera and the router. I am sure the IP and port values are correct. Should I install the rpm version with "debuginfo". It came out of the compilation as well.
I would appreciate your help.
I am enclosing the compiled rpm (PNG added at the end - just rename without it) in case you want to look at it.
[image]

jplx, 2015-11-16:
Additional info: LOG_DEBUG turned on at level 9. I double checked that IP and Port are corect.
11/15/15 23:01:43.859267 zmc_m4[8697].DB9-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2270 [718 - 0: min_y: 0
, max_y:719, min_x:0.00, 1/m:-0.00]
11/15/15 23:01:43.859284 zmc_m4[8697].DB9-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2270 [718 - 1: min_y: 0, max_y:719, min_x:1279.00, 1/m:0.00]
11/15/15 23:01:43.863391 zmc_m4[8697].DB9-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2320 [Deleting active_edge]
11/15/15 23:01:43.863469 zmc_m4[8697].DB9-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2320 [Deleting active_edge]
11/15/15 23:01:43.871042 zmc_m4[8697].DB1-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_monitor.cpp/2774 [Loaded monitor 4(FI9821P), 1 zones]
11/15/15 23:01:43.871123 zmc_m4[8697].INF-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zmc.cpp/195 [Starting Capture]
11/15/15 23:01:43.944356 zmc_m4[8697].INF-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/98 [Priming capture from rtsp://jplx:my_password@192.168.1.28:88/VideoMain]
11/15/15 23:01:44.000240 zmc_m4[8697].DB2-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/215 [OpenFfmpeg called.]
11/15/15 23:01:44.000413 zmc_m4[8697].DB2-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/237 [Number of Options: 2]
11/15/15 23:01:44.000476 zmc_m4[8697].DB2-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/245 [set option 1 'rtsp_transport' to 'tcp']
11/15/15 23:01:44.000494 zmc_m4[8697].DB1-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/254 [Calling avformat_open_input]
11/15/15 23:01:44.025637 zmc_m4[8697].ERR-/home/jplx/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/264 [Unable to open input rtsp://jplx:my_password@192.168.1.28:88/VideoMain due to: Broken pipe]

jplx, 2015-11-18:
I noticed you added Fedora 23 to the zmrepo. So. I first verified that all my files are up to date with the ones listed. They were all installed with the correct revision except ffmpeg files. So, I updated the ffmpeg files (including devel and libs) to 2.8.2-1. I installed your zoneminder binaries and performed the upgrade procedure (zm.conf, etc...) I can see the video by using remote HTTP but not FFMPEG rtsp. I also tried the setup shown by knniggett in issue #734 (remote with rtsp) and it did not work eiher. Here follows the log (level 9) around the failure with FFMPEG rtsp:
11/17/15 22:35:39.933373 zmc_m7[3812].DB9-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2270 [718 - 0: min_y: 0, max_y:719, min_x:0.00, 1/m:-0.00]
11/17/15 22:35:39.933389 zmc_m7[3812].DB9-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2270 [718 - 1: min_y: 0, max_y:719, min_x:1279.00, 1/m:0.00]
11/17/15 22:35:39.933442 zmc_m7[3812].DB9-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2320 [Deleting active_edge]
11/17/15 22:35:39.933470 zmc_m7[3812].DB9-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_image.cpp/2320 [Deleting active_edge]
11/17/15 22:35:39.950778 zmc_m7[3812].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_monitor.cpp/2774 [Loaded monitor 7(FI9821P_3), 1 zones]
11/17/15 22:35:39.950883 zmc_m7[3812].INF-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zmc.cpp/195 [Starting Capture]
11/17/15 22:35:40.035679 zmc_m7[3812].INF-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/98 [Priming capture from rtsp://jplx:mypwd@192.168.1.28:88/VideoMain]
11/17/15 22:35:40.082602 zmc_m7[3812].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/215 [OpenFfmpeg called.]
11/17/15 22:35:40.082660 zmc_m7[3812].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/237 [Number of Options: 2]
11/17/15 22:35:40.082733 zmc_m7[3812].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/245 [set option 1 'rtsp_transport' to 'tcp']
11/17/15 22:35:40.082750 zmc_m7[3812].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/254 [Calling avformat_open_input]
11/17/15 22:35:40.131334 zmc_m7[3812].ERR-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/264 [Unable to open input rtsp://jplx:mypwd@192.168.1.28:88/VideoMain due to: Broken pipe]
11/17/15 22:35:40.171789 zmc_m7[3812].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/361 [ReopenFfmpeg called.]
11/17/15 22:35:40.174994 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/374 [CloseFfmpeg called.]
11/17/15 22:35:40.175128 zmc_m7[3818].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/434 [Sleeping 10 seconds before reopening stream.]
11/17/15 22:35:50.175886 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/215 [OpenFfmpeg called.]
11/17/15 22:35:50.176005 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/237 [Number of Options: 2]
11/17/15 22:35:50.176036 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/245 [set option 1 'rtsp_transport' to 'tcp']
11/17/15 22:35:50.176052 zmc_m7[3818].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/254 [Calling avformat_open_input]
11/17/15 22:35:50.207972 zmc_m7[3818].ERR-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/264 [Unable to open input rtsp://jplx:mypwd@192.168.1.28:88/VideoMain due to: Broken pipe]
11/17/15 22:35:50.289043 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/374 [CloseFfmpeg called.]
11/17/15 22:35:50.289097 zmc_m7[3818].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/434 [Sleeping 10 seconds before reopening stream.]
11/17/15 22:35:52.347943 zmc_m7[3812].INF-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_monitor.cpp/2907 [FI9821P_3: 1000 - Capturing at 76.92 fps]
11/17/15 22:36:00.289211 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/215 [OpenFfmpeg called.]
11/17/15 22:36:00.289324 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/237 [Number of Options: 2]
11/17/15 22:36:00.289357 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/245 [set option 1 'rtsp_transport' to 'tcp']
11/17/15 22:36:00.289374 zmc_m7[3818].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/254 [Calling avformat_open_input]
11/17/15 22:36:00.315595 zmc_m7[3818].ERR-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/264 [Unable to open input rtsp://jplx:mypwd@192.168.1.28:88/VideoMain due to: Broken pipe]
11/17/15 22:36:00.372653 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/374 [CloseFfmpeg called.]
11/17/15 22:36:00.372714 zmc_m7[3818].DB1-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/434 [Sleeping 10 seconds before reopening stream.]
11/17/15 22:36:07.872041 zmc_m7[3812].INF-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_monitor.cpp/2907 [FI9821P_3: 2000 - Capturing at 66.67 fps]
11/17/15 22:36:10.372925 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/215 [OpenFfmpeg called.]
11/17/15 22:36:10.373053 zmc_m7[3818].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/237 [Number of Options: 2]
[... cut ...]
11/17/15 22:36:17.016517 zmc_m7[3812].DB2-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_ffmpeg_camera.cpp/374 [CloseFfmpeg called.]
11/17/15 22:36:17.125973 zmc_m7[3812].INF-/builddir/build/BUILD/ZoneMinder-1.28.1/src/zm_logger.cpp/235 [Terminating Logger]
It shows that for some reason avformat_open_input cannot connect the video stream. The camera I use is brand new: FI9821P V2 rev 2.2 and the firmwares are current.
Do you have any clue as to why avformat_open_input does not connect to the video stream?
May I suggest that you reopen this case?

knnniggett, 2015-11-18:
You have not described a bug, but a camera configuration error.
That camera model is the one I use in my test environments because it has always worked well with ZoneMinder.

We will be glad to provide help, just not in our github forum. Please refer to our posting rules and then create a post in our user forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

In that post you should state your complete camera configuration.

Note that Fedora 23 hosts a copy of ZoneMinder 1.28.1 in their repo. This is new. You need to make sure you don't get their package.  Add an "exclude" statement into the base repo config file so their package gets ignored.

First time users should *not* add any of zoneminder dependencies, such as ffmpeg or vlc, themselves. Instead, let zmrepo install it for you.