ZoneMinder/zoneminder #1159: Migrate www to new host

kylejohnson, 2015-11-12:
I worked on this tonight, and I believe it is finished.

Please report any issues here!

Biggest thing to look out for are redirects and broken links.

E.g. both of these should redirect from the first to the last URL.
```
http://www.zoneminder.com/forums/viewtopic.php?f=32&t=23073
TO
https://www.zoneminder.com/forums/viewtopic.php?f=32&t=23073
TO
https://forums.zoneminder.com/viewtopic.php?f=32&t=23073

AND

http://www.zoneminder.com/wiki/Index.php/Documentation
TO
https://www.zoneminder.com/wiki/Index.php/Documentation
TO
https://wiki.zoneminder.com/Index.php/Documentation
```

knnniggett, 2015-11-12:
Thanks @kylejohnson I'll keep an eye out.

mnoorenberghe, 2017-03-18:
Given that this happened over a year ago I guess it can be closed.

(I'm triaging issues looking for ways to help.)

closed 2017-03-18 by SteveGilvarry