ZoneMinder/zoneminder #1161: HTTP auth failing

imnoboist, 2015-11-13:
I have an AXIS P3214-V and ZM only reports the following logs when trying to connect to it with HTTP:
`Nov 13 11:20:38 shadow zmc_m2[37463]: ERR [Failed authentication: ]`
`Nov 13 11:20:38 shadow zmc_m2[37463]: ERR [Unable to get response]`

... and I get a blue screen when trying to view the feed.

When connecting with wget, browser, and VLC, the connections all work fine.

I'm running version 1.28.1 on Ubuntu 14.04 with all the latest updates.

Let me know what info you need from me to help. I'm a total newb to ZM but very excited at the possibilities!

Thanks!

-Sean

imnoboist, 2015-11-17:
I've been trying to diagnose this issue myself and I've come up with the following logs after turning on debugging:

`Nov 13 12:26:29 shadow zmc_m2[2011]: DB2 [HA1 pre-md5: viewonly:AXIS_ACCC8E45256A:password]`
`Nov 13 12:26:29 shadow zmc_m2[2011]: DB2 [HA2 pre-md5: GET:/mjpg/video.mjpg]`
`Nov 13 12:26:29 shadow zmc_m2[2011]: DB2 [pre-md5: b8ae326e9e1e97672b2e5b83256fc528:ulLY33EkBQA:00000001:0a4f113b:auth:ade69d1be2b68919b97f126202cc49ad]`
`Nov 13 12:26:59 shadow zmc_m2[2011]: DB3 [HTTP/1.1 401 Unauthorized#015#012Date: Fri, 13 Nov 2015 20:26:59 GMT#015#012Server: Apache/2.4.10 (Unix) OpenSSL/0.9.8zf#015#012WWW-Authenticate: Digest realm="AXIS_ACCC8E45256A", nonce="BQ2m4XEkBQA=edae1cf866638759b9f17d71de0f3a97fbced131", algorithm=MD5, stale=true, qop="auth"#015#012Content-Length: 381#015#012Keep-Alive: timeout=5, max=100#015#012Connection: Keep-Alive#015#012Content-Type: text/html; charset=iso-8859-1#015#012#015#012<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">#012<html><head>#012<title>401 Unauthorized</title>#012</head><body>#012<h1>Unauthorized</h1>#012<p>This server could not verify that you#012are authorized to access the document#012requested.  Either you supplied the wrong#012credentials (e.g., bad password), or your#012browser doesn't understand how to supply#012the credentials required.</p>#012</body></html>#012 Unauthorized]`
`Nov 13 12:27:02 shadow zmc_m2[2011]: DB2 [Sending request: GET /mjpg/video.mjpg HTTP/1.0#015#012User-Agent: ZoneMinder/1.28.1#015#012Host: 10.0.32.2#015#012Connection: Keep-Alive#015#012Authorization: Digest username="viewonly", realm="AXIS_ACCC8E45256A", nonce="ulLY33EkBQA", uri="/mjpg/video.mjpg", qop=auth, nc=00000001, cnonce=0a4f113b, response="671972a69a42a2ffe1f64090a9deadb9"#015#012#015#012]`

I have meticulously copied and pasted the correct settings from VLC into the source tab of ZM and still getting the same errors.

Thanks again for any help!

-Sean

pliablepixels, 2015-11-17:
Hi Sean, there are many users using Axis - it may be a good idea for you to post the question in the ZM forums at https://forums.zoneminder.com/ 

When posting, its a good idea to also include a picture of what setting you specified in the ZM source tab settings for that camera

knnniggett, 2015-11-18:
Thank you for posting this into our user forum. 
For future requests, please refer to our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-11-18 by knnniggett